Sloeber/arduino-eclipse-plugin #952: V4 product for MAC installs with erorrs

cbwarren, 2018-04-05:
This installs but has a list of errors in error log; unable to open new sketch as it complains it can't find boards.txt.
[image]

Suggestions?

jantje, 2018-04-05:
Did you tell your firewall that Sloeber is allowed access to the network?
Did you follow these instructions http://eclipse.baeyens.it/installAdvice.shtml

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2018-04-09:
I'll assume this issue can be closed

closed 2018-04-09 by jantje

cbwarren, 2018-04-12:
I tried again with my firewall off;
I get the following in a popup window before any installation starts:

The indexer option "index source files not included in the build" must be
off in windows->preferences->C/C++->indexer
and
The indexer option "index unused headers" must be off in
windows->preferences->C/C++->indexer

Sloeber might still function but if you get strange results you know where
to look.
Do not create an issue if you see this!!!

The plugin Needs write access to
/private/var/folders/hw/wczzy0j15p77fwp86349230c0000gp/T/AppTranslocation/252A210F-09C3-40A3-AE5E-0B689DBE7D3E/d/sloeber.app/Contents/Eclipse/arduinoPlugin

   1. I did do the instructions! - i wouldn't have raised issue if I had
   not :-)
   2. indexer option message is for windows(?) but i've downloaded the MAC
   version? This sounds like an option in the installer, but where do I set
   these options?
   3. re the firewall: it is not enabled; I note that some files are being
   downloaded during the install but these are showing up in the error log; if
   it was a firewall issue, why would some files download and not others?
   4. re the message "The plugin Needs write access to
   ..../Eclipse/arduinoPlugin": I can't find this file on my computer, so I
   can't change the write access???

Appreciate any insight you can provide.

Regards,

 Bruce Warren

garyrice, 2018-06-05:
I just downloaded Sloeber (full) v4.2 (Mac). I started the app and received this same error. Is the message "you may have strange behavior" real or can I just ignore it?

Gary