Sloeber/arduino-eclipse-plugin #953: Communication port unavailable.

phillip-toone, 2018-04-06:
I am attempting to use Sloeber to upload the blink example on Ubuntu 16.04.  I download Slober [here](https://github.com/Sloeber/arduino-eclipse-plugin/releases/download/4_2/V4.2_linux64.2017-11-20_02-16-52.tar.gz), extract the file, and execute the sloeber-ide file.  Slober starts with the following error message which I ignore.
```
The indexer option "index source files not included in the build" must be off in windows->preferences->C/C++->indexer 
and
The indexer option "index unused headers" must be off in windows->preferences->C/C++->indexer 

Sloeber might still function but if you get strange results you know where to look.
Do not create an issue if you see this!!!
```
I create a new project (File > New > Arduino Sketch).  I give my project and name and click next.  I attempt to assign a port to this project but there are no options listed in the drop down menu.  I issue the lsusb command from the command line and see my Arduino Uno listed as:
```
Bus 010 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
```
I click next and choose the blink example (examples > 01.Basics > Blink) from the next screen.  I finish the process by clicking Finish.  I see a Blink.ino file is listed in the project tree.  I attempt to upload this sketch (Arduino > Upload Sketch).  I choose "Yes" when asked "Build before upload".  I get the following results in the Console.
```
Starting upload
using arduino loader
Com port not found. Skipping reset

Launching/home/ptoone/Downloads/sloeber//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -C/home/ptoone/Downloads/sloeber//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega328p -carduino -P -b115200 -D -Uflash:w:/home/ptoone/Documents/sloeber-workspace/issue/Release/issue.hex:i 
Output:
avrdude: ser_open(): can't open device "-b115200": No such file or directory

avrdude done.  Thank you.

avrdude finished
upload done
```
I open the "Serial monitor view" but again there are no options listed in the drop down menu for port assignment.  I click the "+" to add a serial port to the monitor.  This opens a window with the option to choose a "Serial port to connect to:" but again the drop down menu is empty.

How do I assign a port to use to upload a sketch?

jantje, 2018-04-06:
Project properties->arduino->port is a combotext field so if the port is not discovered you can type it yourself.
If the project is selected and you select the toolbar action "open serial monitor" Sloeber will use the  serial port from project properties->arduino->port and the baudrate as you specified in Serial->begin in your sketch.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2018-04-09:
I'll assume this issue can be closed

closed 2018-04-09 by jantje

phillip-toone, 2018-04-10:
Thank you for your response and sorry for my delayed response.  I have figured out why there are no options listed in the drop down menu.  I am running Ubuntu and did not have my computer set to grant me as a user access to these ports.  The easiest way to fix this is to install the Arduino IDE and let it configure this for you.  When you first start up the Arduino IDE it will check to see if you have access to the port.  If you don't it will offer to fix this for you but will required a sudo password.  After providing the password the Arduino IDE will inform you that you need to log off and back in for these changes to take effect.  When you log back in the port will be listed in the drop down menu of Sloeber.

This is simple workaround but Sloeber would be a more refined product if this feature were included.

jantje, 2018-04-10:
>This is simple workaround but Sloeber would be a more refined product if this feature were included.

It wouldn't be free anymore as well.

phillip-toone, 2018-04-10:
> It wouldn't be free anymore as well.

I'm not sure we understand one another. I'm not suggesting that you incorporate Arduino IDE into Sloeber.  I'm only suggesting that a feature, similar to what is found in the Arduino IDE, be added to Sloeber.  I would love to incorporate this feature into Sloeber but I suspect my programming skills for this are not up to the job.

phillip-toone, 2018-04-10:
I understand. I was just thinking this would be a nice feature to add to a prioritized list of things to do someday.