Sloeber/arduino-eclipse-plugin #955: Board Configuration Properties Don't scroll

coogle, 2018-04-08:
I am running the plugin on a screen that doesn't have the best resolution. When I try to go into the properties for a project and set the board (in my case, a NodeMCU 1.0) there are a number of configuration options (80/160mhz, flash size, etc.) I need to populate in order to actually set the board and apply the settings.

Unfortunately, this view doesn't actually allow you to scroll vertically. This means at least one or two of those required configuration settings are off-window and I have to basically blind set them using tab/arrow keys just to get the board settings to apply.

Any chance this could be corrected and a vertical scroll be added?

jantje, 2018-04-08:
@coogle 
This is indeed a issue. the workaround is to double click the titlebar to maximize the screen.  At that point there should be a scrollbar.
SWT is not my strongest part so I hope someone jumps in to help fix this.

Sidru, 2018-09-18:
Did anyone find a solution? I have the same problem...

jantje, 2018-09-18:
@Sidru 
except for the workaround described in my previous post, no

jantje, 2018-11-13:
@rlogiacco @wimjongman 
Can you guys take a look?
I really need someone with SWT knowledge to fix this issue. I created a branch with the current state.
The issue is simple. When changing from arduino uno to esp8266 generic board there are far more options in the second one.
These options are likely not visible and no scrollbar appears.
After giving up on getting the  existing CDT controlled scrollbar to work I tried to add a scrollbar myself.
The scrollbar is now there however it is always there and it doesn't scroll :disappointed:
I need help (pun intended)

jantje, 2018-11-23:
Thanks @wimjongman I would never have been able tofix this on my own :-)

closed 2018-11-23 by jantje

Sidru, 2018-11-25:
@jantje  and @wimjongman  thanks for your work!

jantje, 2019-02-10:
Seems there is still an issue when switching boards inside a boards.txt file (like with esp32)