Sloeber/arduino-eclipse-plugin #957: avrdude options wrong

phillip-toone, 2018-04-10:
I am unable to upload a sketch with Sloeber.  When I examine the call to avrdude:
```
Launching/home/ptoone/Downloads/sloeber//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -C/home/ptoone/Downloads/sloeber//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega328p -cstk500v1 -P/dev/ttyUSB2 -b19200 -Uflash:w:/home/ptoone/Documents/sloeber-workspace/qqq/Release/qqq.hex:i 
```
I realize that programmer option (-c stk500v1) and the baudrate option (-b 19200) are both wrong.  Modifying these option to (-c arduino) and (-b 57600) work when executed through the command line.  Is there I way I should be able to change these option within Sloeber?

jantje, 2018-04-10:
Which board are you using? esp8266? #907

phillip-toone, 2018-04-10:
I am just attempting to upload the blink example to an Arduino Nano.  I see the link to issue #907 and realize that I am not presented with all of these options, only Board, Upload Protocol, Port, and Processor.

jantje, 2018-04-10:
I tried here with the nano and the command seems fine to me`
LaunchingC:\eclipse\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -CC:\eclipse\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega328p -carduino -Pnone -b115200 -D -Uflash:w:L:\libExamples\01173_Library_ConfigManager/Release/01173_Library_ConfigManager.hex:i `

[image]

phillip-toone, 2018-04-10:
Okay, I see that you are using "Default" as your "Upload Protocol". I was using "Arduino as ISP" because that is what works in the Arduino IDE.  But when I change the "Upload Protocol" to "default" the baud rate option is set to "-b 115200" and this doesn't work unless it is "-b 57600".

Out of curiosity I tried uploading the blink example to an Arduino Mega with the "Upload Protocol" set to "default" and it uploaded without a problem.  The baud option is set to "-b 115200" and that seems to work with the Mega.

I believe the baud option for the Nano needs to be changed from "-b 115200" to "-b 57600".  This is how the Arduino IDE is set up to program the Nano and it is what works on the hardware I have tested.

jantje, 2018-04-10:
I tried with arduino IDE
`C:\Users\jan\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/bin/avrdude -CC:\Users\jan\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf -v -patmega328p -carduino -PCOM8 -b115200 -D -Uflash:w:C:\Users\jan\AppData\Local\Temp\arduino_build_918186/sketch_apr11a.ino.hex:i `

closed 2018-04-10 by jantje

felichsm, 2018-05-15:
Even though this Issue is closed, I wanted to let you know I encountered the same Problem. 
My Project > Arduino Settings are as can be seen in the Picture:
[image]
Which leads to the following upload command:
_D:\sloeber\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -CD:\sloeber\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega328p -carduino -PCOM3 -b115200 -D -Uflash:w:C:\Users\felix\Documents\sloeber-workspace\TestSketch/Release/TestSketch.hex:i_ 
The Upload however fails with:
_avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x55_
where _resp=0x55_ is not necesarrily always _0x55_ but also other response codes like _0x75_ etc
Since my **Arduino IDE** uses an Upload speed of 57600 Baud for the Nano, I configured the upload Baudrate for Sloeber the same by editing the _sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.21\boards.txt_:
[image]
**After reapplying the Project > Arduino settings, the upload now works.** 
I have to admit that my Arduino Nano is a chinese clone and that this might be the reason, only baudrate 57600 works for me.   
I wanted to share my experience in case anyone runs into the same problem.
Also I want to suggest, a Dropdown-Option for selecting the upload baudrate might be a nice feature 

Thank you Jantje for the great work developing Sloeber!!!

jantje, 2018-05-15:
@felichsm 
Your problem has nothing to do with Sloeber as -per your own workaround- sloeber uses the baudrate as specified in boards.txt.
IMHO your arduino ide has avr platform 1.6.20 installed where sloeber has 1.6.21
1.6.20 states
`nano.menu.cpu.atmega328.upload.speed=57600`
1.6.21 states
`nano.menu.cpu.atmega328.upload.speed=115200`
If you would upgrade your avr platform in Arduino ide to 1.6.21 the Arduino IDE would use 115200.