Sloeber/arduino-eclipse-plugin #960: ARDUINO_BOARD symbol build error

danielfmo, 2018-04-12:
I'm getting this issue again in Sloeber 4.2.0.201804050109 and ESP 2.4.0
Does anyone else confirm the issue?
It works in Arduino IDE 1.8.5

C:\Users...\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS\ESP8266mDNS.cpp: In member function 'void MDNSResponder::enableArduino(uint16_t, bool)':
:0:15: error: 'ESP8266_NODEMCU' was not declared in this scope
C:\Users...\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS\ESP8266mDNS.cpp:902:44: note: in expansion of macro 'ARDUINO_BOARD'
addServiceTxt("arduino", "tcp", "board", ARDUINO_BOARD);
^
libraries\ESP8266mDNS\subdir.mk:18: recipe for target 'libraries\ESP8266mDNS\ESP8266mDNS.cpp.o' failed
make: *** [libraries\ESP8266mDNS\ESP8266mDNS.cpp.o] Error 1

jantje, 2018-04-19:
The fix is done at install time.
Please remove the platform and reinstall it.

dipeshkumar1998, 2018-04-21:
Hey, I have the same problem.
I am using the Sublime text editor and Stino(Arduino like IDE) added to it.
I have independently downloaded the Esp8266 packages. and added to the Stino.