Sloeber/arduino-eclipse-plugin #963: Not possible to update board/platform after update sloeber

hreintke, 2018-04-17:
I started using a newer version of ArduinoPlugin (4.0 -> Nighly)

Old environment : 

Sloeber 4.0
- AVR 1.6.18
- ESP8266 2.3.0 & git
- ESP32
- One workspace with all projects, using all of the above 

Did a fresh install : 

Sloeber Nightly
- AVR 1.16.21
- ESP 2.4.1 & git
- Using the same workspace as before

I see all projects in eclipse, the ones using a platform not available in new -> no Board is mentioned.

When trying to update properties to set the right board,(selecting properties_.arduino) 
I get the popup : 
[image]

Getting multiple popups after that, stating 
- Current page contains invalid values
- nullpointer exception, 
Sloeber itself doesn't crash.

Only option to get past this is installing older versions of platforms and then updating board.
Would expect that it is always possible to select a platform for a project.

jantje, 2018-04-17:
It always works for me after a couple of tries.

hreintke, 2018-04-17:
If you mean just trying to update the properties : It doesn't for me after 10 times.
Even then it's a bad user experience, of course up to you to consider it a bug or not.

jantje, 2018-04-17:
it is a bug/issue for me.
I just stated that I got out of this by retrying which is less work than installing "older versions.
But if I understood correctly your usecase is differently as you moved eclipse location as well as from stable to nightly (I go from nightly to nightly)
I consider this to be fixed when the next stable is being made. Each nightly may behave differently and it is to much work to test/fix for each and every nightly. I guess  I see this as part of the (b)leeding edge experience.
If you have stack traces for the null pointer exceptions, that would be a great help.

jantje, 2018-04-17:
There is 1 nullpointer exception in the log :-)
Note that the nightly has serious changes internally so this kind of issues are considered normal due to the transition.
Not having to go through these is a benefit of staying with the stable version.

hreintke, 2018-04-17:
Yes, only one nullpointer exception. I made the log special for  you with only the exception from the issue.
And I do know that a nightly or git build has risks.
For my usecase however I prefer to have the upload speed issue fixed.

And in the meantime I can help you in cleaning/improving the quality of the next stable.

hreintke, 2018-04-18:
For me, I understand the issue and do have a workaround for it.
Do you want to keep it open for you reference or shall i close it.

PS, FYI : 
I opened an issue on esp32 github on failing compile. I've seen a simular before but can't remember where. https://github.com/espressif/arduino-esp32/issues/1327

jantje, 2018-04-18:
Keep it open to fix the nullpointer.

jantje, 2018-06-19:
I can't remember. Are all issues closed here now?

closed 2018-08-24 by jantje