Sloeber/arduino-eclipse-plugin #965: make 'clean' target fails

DrObscure, 2018-04-18:
Most recent 'nightly' update -- the one that installed all the GNU pieces -- is now failing in the auto generated makefile for the 'clean' target:
---
<
17:12:01 **** Clean-only build of configuration Release for project DCCpp_Uno ****
"C:\\Sloeber\\Eclipse\\neon\\arduinoPlugin\\tools\\make\\make" clean 

del   ./core\core\WInterrupts.c.d ./core\core\hooks.c.d ./core\core\wiring.c.d ./core\core\wiring_analog.c.d ./core\core\wiring_digital.c.d ./core\core\wiring_pulse.c.d ./core\core\wiring_shift.c.d  C:\Projects\DCC++\BaseStation-master\DCCpp_Uno\Release\arduino.ar  ./core\core\CDC.cpp.o ./core\core\HardwareSerial.cpp.o ./core\core\HardwareSerial0.cpp.o ./core\core\HardwareSerial1.cpp.o ./core\core\HardwareSerial2.cpp.o ./core\core\HardwareSerial3.cpp.o ./core\core\IPAddress.cpp.o ./core\core\PluggableUSB.cpp.o ./core\core\Print.cpp.o ./core\core\Stream.cpp.o ./core\core\Tone.cpp.o ./core\core\USBCore.cpp.o ./core\core\WInterrupts.c.o ./core\core\WMath.cpp.o ./core\core\WString.cpp.o ./core\core\abi.cpp.o ./core\core\hooks.c.o ./core\core\main.cpp.o ./core\core\new.cpp.o ./core\core\wiring.c.o ./core\core\wiring_analog.c.o ./core\core\wiring_digital.c.o ./core\core\wiring_pulse.S.o ./core\core\wiring_pulse.c.o ./core\core\wiring_shift.c.o  ./Accessories.cpp.o ./CurrentMonitor.cpp.o ./DCCpp_Uno.cpp.o ./EEStore.cpp.o ./Outputs.cpp.o ./PacketRegister.cpp.o ./Sensor.cpp.o ./SerialCommand.cpp.o  DCCpp_Uno.hex  ./core\core\wiring_pulse.S.d  DCCpp_Uno.elf  ./Accessories.cpp.d ./CurrentMonitor.cpp.d ./DCCpp_Uno.cpp.d ./EEStore.cpp.d ./Outputs.cpp.d ./PacketRegister.cpp.d ./Sensor.cpp.d ./SerialCommand.cpp.d  ./core\core\CDC.cpp.d ./core\core\HardwareSerial.cpp.d ./core\core\HardwareSerial0.cpp.d ./core\core\HardwareSerial1.cpp.d ./core\core\HardwareSerial2.cpp.d ./core\core\HardwareSerial3.cpp.d ./core\core\IPAddress.cpp.d ./core\core\PluggableUSB.cpp.d ./core\core\Print.cpp.d ./core\core\Stream.cpp.d ./core\core\Tone.cpp.d ./core\core\USBCore.cpp.d ./core\core\WMath.cpp.d ./core\core\WString.cpp.d ./core\core\abi.cpp.d ./core\core\main.cpp.d ./core\core\new.cpp.d 

**Invalid switch - "core\core\WInterrupts.c.d".
makefile:93: recipe for target 'clean' failed
make: [clean] Error 1 (ignored)**
' '

17:12:01 Build Finished (took 303ms)
>

jantje, 2018-04-18:
Indeed.
Thanks for the heads up. Should be fixed tomorrow.

DrObscure, 2018-04-19:
Thanks ..  all better now :)

closed 2018-04-19 by DrObscure