Sloeber/arduino-eclipse-plugin #967: Fresh install compilation error.

miky2k, 2018-04-19:
'''
09:55:34 **** Incremental Build of configuration Release for project test ****
make all 
Building file: /home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp
Starting C++ compile
"/home/users/bin/sloeber//arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/users/bin/sloeber//arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/include" "-I/home/users/bin/sloeber//arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/lwip2/include" "-I/home/users/bin/sloeber//arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-I/home/users/workspace/test/Release/core" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=1460  -DARDUINO=10802 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 '-DARDUINO_BOARD="ESP8266_GENERIC"' -DLED_BUILTIN=0 -DF_CRYSTAL=40000000 -DESP8266  -I"/home/users/bin/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266" -I"/home/users/bin/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/variants/generic" -I"/home/users/bin/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer" -I"/home/users/bin/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WebServer/src" -I"/home/users/bin/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266mDNS" -I"/home/users/bin/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi" -I"/home/users/bin/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src" -I"/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6" -I"/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src" -I"/home/users/bin/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/SPI" -MMD -MP -MF"libraries/WiFi/src/WiFiUdp.cpp.d" -MT"libraries/WiFi/src/WiFiUdp.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp"  -o  "libraries/WiFi/src/WiFiUdp.cpp.o"
In file included from /home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:29:0:
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.h:25:0: warning: "UDP_TX_PACKET_MAX_SIZE" redefined [enabled by default]
 #define UDP_TX_PACKET_MAX_SIZE 24
 ^
In file included from /home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/utility/wifi_drv.h:26:0,
                 from /home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:26:
/home/users/bin/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/WiFiUdp.h:28:0: note: this is the location of the previous definition
 #define UDP_TX_PACKET_MAX_SIZE 8192
 ^
In file included from /home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:29:0:
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.h:27:7: error: redefinition of 'class WiFiUDP'
 class WiFiUDP : public UDP {
       ^
In file included from /home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/utility/wifi_drv.h:26:0,
                 from /home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:26:
/home/users/bin/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/libraries/ESP8266WiFi/src/WiFiUdp.h:32:7: error: previous definition of 'class WiFiUDP'
 class WiFiUDP : public UDP, public SList<WiFiUDP> {
       ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp: In constructor 'WiFiUDP::WiFiUDP()':
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:35:22: error: class 'WiFiUDP' does not have any field named '_sock'
 WiFiUDP::WiFiUDP() : _sock(NO_SOCKET_AVAIL) {}
                      ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp: In member function 'virtual uint8_t WiFiUDP::begin(uint16_t)':
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:45:9: error: '_sock' was not declared in this scope
         _sock = sock;
         ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:46:9: error: '_port' was not declared in this scope
         _port = port;
         ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::available()':
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:56:7: error: '_sock' was not declared in this scope
   if (_sock != NO_SOCKET_AVAIL)
       ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp: In member function 'virtual void WiFiUDP::stop()':
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:66:8: error: '_sock' was not declared in this scope
    if (_sock == NO_SOCKET_AVAIL)
        ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:69:26: error: '_sock' was not declared in this scope
    ServerDrv::stopClient(_sock);
                          ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::beginPacket(IPAddress, uint16_t)':
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:88:7: error: '_sock' was not declared in this scope
   if (_sock == NO_SOCKET_AVAIL)
       ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:90:7: error: '_sock' was not declared in this scope
   if (_sock != NO_SOCKET_AVAIL)
       ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::endPacket()':
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:101:32: error: '_sock' was not declared in this scope
libraries/WiFi/src/subdir.mk:48: set di istruzioni per l'obiettivo "libraries/WiFi/src/WiFiUdp.cpp.o" non riuscito
  return ServerDrv::sendUdpData(_sock);
                                ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp: In member function 'virtual size_t WiFiUDP::write(const uint8_t*, size_t)':
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:111:27: error: '_sock' was not declared in this scope
  ServerDrv::insertDataBuf(_sock, buffer, size);
                           ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::read()':
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:125:23: error: '_sock' was not declared in this scope
    ServerDrv::getData(_sock, &b);
                       ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::read(unsigned char*, size_t)':
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:137:31: error: '_sock' was not declared in this scope
    if (!ServerDrv::getDataBuf(_sock, buffer, &size))
                               ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp: At global scope:
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:132:5: warning: unused parameter 'len' [-Wunused-parameter]
 int WiFiUDP::read(unsigned char* buffer, size_t len)
     ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::peek()':
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:152:22: error: '_sock' was not declared in this scope
   ServerDrv::getData(_sock, &b, 1);
                      ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp: In member function 'virtual IPAddress WiFiUDP::remoteIP()':
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:167:25: error: '_sock' was not declared in this scope
  WiFiDrv::getRemoteData(_sock, _remoteIp, _remotePort);
                         ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp: In member function 'virtual uint16_t WiFiUDP::remotePort()':
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:177:25: error: '_sock' was not declared in this scope
  WiFiDrv::getRemoteData(_sock, _remoteIp, _remotePort);
                         ^
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp: In member function 'virtual int WiFiUDP::endPacket()':
/home/users/bin/sloeber/arduinoPlugin/libraries/WiFi/1.2.6/src/WiFiUdp.cpp:102:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
make: *** [libraries/WiFi/src/WiFiUdp.cpp.o] Errore 1

09:55:34 Build Finished (took 719ms)
'''

JayNaire, 2018-04-22:
Me too. Exactly the same behaviour with WifiUDP, fresh install  Sloeber, the Eclipse Arduino IDE	4.2.0.201711200109, Linux Mint 18

ATho95, 2018-06-12:
It seems that creating a new sketch (using the ArduinoOTA example OTALeds.ino) creates two conflicting linked folders ('WiFi' and 'ESP8266WiFi') in the libraries directory. Also Include Path entries for both libs are made.
I got rid of the redefinition error by deleting the WiFi link in the libraries directory and the Include Path entries pointing to the WiFi lib.

Btw, when you create a new sketch using the WiFiWebServer example, only the WiFi entries are generated. So I assume the #include <ESP8266WiFi.h> in the OTA example is the trigger to generate the links to ESP8266WiFi lib.

closed 2018-09-08 by jantje