Sloeber/arduino-eclipse-plugin #969: Archive fails whene updating Core only

hreintke, 2018-04-24:
When working on Core the archive fails.
Manual remove of the arduino.ar file let's it archive correctly

'''
'Starting archiver'
"C:\Utilities\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "C:/Userdata/ESP8266/Projects/WebServer/Release/arduino.ar"    C:/Userdata/ESP8266/Projects/WebServer/Release/arduino.ar  .\core\core\umm_malloc\umm_malloc.c.o  .\core\core\spiffs\spiffs_cache.c.o .\core\core\spiffs\spiffs_check.c.o .\core\core\spiffs\spiffs_gc.c.o .\core\core\spiffs\spiffs_hydrogen.c.o .\core\core\spiffs\spiffs_nucleus.c.o  .\core\core\libb64\cdecode.c.o .\core\core\libb64\cencode.c.o  .\core\core\Esp-version.cpp.o .\core\core\Esp.cpp.o .\core\core\FS.cpp.o .\core\core\FunctionalInterrupt.cpp.o .\core\core\HardwareSerial.cpp.o .\core\core\IPAddress.cpp.o .\core\core\MD5Builder.cpp.o .\core\core\Print.cpp.o .\core\core\Schedule.cpp.o .\core\core\Stream.cpp.o .\core\core\StreamString.cpp.o .\core\core\Tone.cpp.o .\core\core\Updater.cpp.o .\core\core\WMath.cpp.o .\core\core\WString.cpp.o .\core\core\abi.cpp.o .\core\core\base64.cpp.o .\core\core\cbuf.cpp.o .\core\core\cont.S.o .\core\core\cont_util.c.o .\core\core\core_esp8266_eboot_command.c.o .\core\core\core_esp8266_flash_utils.c.o .\core\core\core_esp8266_i2s.c.o .\core\core\core_esp8266_main.cpp.o .\core\core\core_esp8266_noniso.c.o .\core\core\core_esp8266_phy.c.o .\core\core\core_esp8266_postmortem.c.o .\core\core\core_esp8266_si2c.c.o .\core\core\core_esp8266_sigma_delta.c.o .\core\core\core_esp8266_timer.c.o .\core\core\core_esp8266_wiring.c.o .\core\core\core_esp8266_wiring_analog.c.o .\core\core\core_esp8266_wiring_digital.c.o .\core\core\core_esp8266_wiring_pulse.c.o .\core\core\core_esp8266_wiring_pwm.c.o .\core\core\core_esp8266_wiring_shift.c.o .\core\core\debug.cpp.o .\core\core\gdb_hooks.c.o .\core\core\heap.c.o .\core\core\libc_replacements.c.o .\core\core\pgmspace.cpp.o .\core\core\sntp-lwip2.c.o .\core\core\spiffs_api.cpp.o .\core\core\spiffs_hal.cpp.o .\core\core\time.c.o .\core\core\uart.c.o 
C:\Utilities\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar: unable to rename 'C:/Userdata/ESP8266/Projects/WebServer/Release/arduino.ar'; reason: File exists
makefile:85: recipe for target 'C:\Userdata\ESP8266\Projects\WebServer\Release\arduino.ar' failed
make: *** [C:\Userdata\ESP8266\Projects\WebServer\Release\arduino.ar] Error 1
'''

jantje, 2018-04-24:
Can you provide steps to recreate this issue?

hreintke, 2018-04-24:
Sure, sorry should have done that in the first description.

ESP8266 project 
- Clean project
- Build project
- Update Core file
- Build project
- Update Core file
- Build project -> fails.

Now running latest nightly but also noticed this in stable

jantje, 2018-04-24:
What exactly do you mean with "Update Core file'? A file in the core folder?

hreintke, 2018-04-24:
Yes, any file in the core/core directory.

[image]

jantje, 2018-04-24:
i can confirm this is indeed happening. No clue why though.

jantje, 2018-04-25:
Anyway a workaround is to replace in platform.txt
`compiler.ar.flags=cru`
with
`compiler.ar.flags=cr`
Note this is only needed if you want to modify the core of a platform.

Some thoughts
I see no way to test this would not happening in Arduino IDE as you can not do changes to the core files in the Arduino IDE.
IMHO it is a gnu gcc bug but I also wonder why the 'u' is needed. my source https://www.lifewire.com/ar-linux-command-4093866

hreintke, 2018-04-25:
Sorry if this remark is nonsense, I have no detailed knowledge of the subject.

In the command link you  provided is : `ar [-X32_64] [-]p[mod [relpos] [count]] archive [member...]`

The executed command is : `"C:\Utilities\sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "C:/Userdata/ESP8266/Projects/WebServer/Release/arduino.ar"    C:/Userdata/ESP8266/Projects/WebServer/Release/arduino.ar  .\core\core\umm_malloc\umm_malloc.c.o .....  `

In that the archive name is included twice. 
When I manually run the ar command with only once the archive name in, it runs OK.

Based on what definition are you building the ar command ?

jantje, 2018-04-25:
>Sorry if this remark is nonsense, I have no detailed knowledge of the subject.

Nor do I :-)
I missed that the ar is 2 times in the command

hreintke, 2018-04-28:
Can you point me to the part of the code where the ar command is generated ?

jantje, 2018-04-28:
project properties->C/C++ build->settings->Arduino archiver->command

you should find `${A.RECIPE.AR.PATTERN.1} ${FLAGS}  ${A.ARCHIVE_FILE_PATH} ${INPUTS} ${A.RECIPE.AR.PATTERN.2} ${A.RECIPE.AR.PATTERN.3} ${JANTJE.EXTRA.ARCHIVE} ${JANTJE.EXTRA.ALL}`

The A. environment variables come from platform.txt and boards.txt the others are from CDT
${FLAGS}  should be empty

The problem is that 
`${A.RECIPE.AR.PATTERN.1 = "${A.COMPILER.PATH}${A.COMPILER.AR.CMD}" ${A.COMPILER.AR.FLAGS} ${A.COMPILER.AR.EXTRA_FLAGS} "${A.BUILD.PATH}/arduino.ar" `

So it contains `"${A.BUILD.PATH}/arduino.ar"`  and then the command is `${A.ARCHIVE_FILE_PATH}`
resulting in having the archive 2 times in the command.

The root cause of that is that the esp8266 platform.txt states:
`recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/arduino.ar" "{object_file}"`
and that should be
`recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}"  "{object_file}"`

hreintke, 2018-05-01:
Trying to really understand the issue. If I want to create an issue on the esp8266 I will need that.

I checked the arduino platform.txt and that contains : 
```
archive_file_path={build.path}/{archive_file}
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}"
```
When the archive files is Arduino.ar, that is the equivalence of the esp8266 platform.txt
```
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/arduino.ar" "{object_file}"
```

And in the environment variables there is also : 
`A.ARCHIVE_FILE = arduino.ar`
`A.ARCHIVE_FILE_PATH = ${A.BUILD.PATH}/${A.ARCHIVE_FILE}`

That then leads to the same ar command.

jantje, 2018-05-01:
It leads to the same result if you assume there is no additional parsing by the tool.
However the commands as in the platform.txt can not be passed on "as-is" to  CDT (other tools have the same issue). Certain key info needs to be "known by the tool".
For the ar command the input is "object files" and the output is a archive file.
Arduino IDE has tagged the "object files" as {object_file} fromt the start of the platform.txt (If I remember correctly).
The archive file has been hardcoded as core.a arduino.a {build.path}/{archive_file}.
I think it now stabilized on {archive_file_path}
Arduino introduced archive_file_path in V1.6.6 the doc was not yet updated (I did this just  now)
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification/_compare/4601bc4542eb69dbbd115df66136424bed038fa3

Note that arduino didn't update the linker in avr which still uses "{build.path}/{archive_file}"  which is really bad.

hreintke, 2018-05-01:
Think I tried to say two things in one sentence, new try.

>The root cause of that is that the esp8266 platform.txt states:
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/arduino.ar" "{object_file}"
and that should be
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}"

And 
>And in the environment variables there is also :
A.ARCHIVE_FILE = arduino.ar
A.ARCHIVE_FILE_PATH = ${A.BUILD.PATH}/${A.ARCHIVE_FILE}

Both recipe.ar.pattern's will evaluate to the same
If that is correct, how can that be the root case of the issue

jantje, 2018-05-01:
because sloeber cuts the command based on keywords like  "{archive_file_path}" and "{object_file}"
here https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/tools/Helpers.java#L829

jantje, 2018-05-11:
any progress?

hreintke, 2018-05-12:
I've been away from home last week. Hope that I will find some time next week.

jantje, 2018-07-15:
I guess this one can be closed now.

closed 2018-09-08 by jantje