Sloeber/arduino-eclipse-plugin #974: "Add library to the Selected project" creates unhandled exeption

Panther2804, 2018-05-02:
When i try to add a Library this error is created.
There is no dialog popping up.
`org.eclipse.e4.core.di.InjectionException: java.lang.NullPointerException
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:65)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invokeUsingClass(InjectorImpl.java:305)
	at org.eclipse.e4.core.internal.di.InjectorImpl.invoke(InjectorImpl.java:239)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.invoke(ContextInjectionFactory.java:132)
	at org.eclipse.e4.core.commands.internal.HandlerServiceHandler.execute(HandlerServiceHandler.java:152)
	at org.eclipse.core.commands.Command.executeWithChecks(Command.java:494)
	at org.eclipse.core.commands.ParameterizedCommand.executeWithChecks(ParameterizedCommand.java:487)
	at org.eclipse.e4.core.commands.internal.HandlerServiceImpl.executeHandler(HandlerServiceImpl.java:210)
	at org.eclipse.e4.ui.workbench.renderers.swt.HandledContributionItem.executeItem(HandledContributionItem.java:431)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.handleWidgetSelection(AbstractContributionItem.java:446)
	at org.eclipse.e4.ui.workbench.renderers.swt.AbstractContributionItem.lambda$2(AbstractContributionItem.java:472)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:86)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4428)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1079)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4238)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3817)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1150)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1039)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:153)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:680)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:594)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:151)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:653)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:590)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1499)
Caused by: java.lang.NullPointerException
	at io.sloeber.core.tools.Libraries.findAllSubFolders(Libraries.java:77)
	at io.sloeber.core.tools.Libraries.findAllPrivateLibraries(Libraries.java:125)
	at io.sloeber.core.tools.Libraries.getAllInstalledLibraries(Libraries.java:198)
	at io.sloeber.core.api.Sketch.getAllAvailableLibraries(Sketch.java:179)
	at io.sloeber.ui.Import_Libraries_Page.createSourceGroup(Import_Libraries_Page.java:85)
	at io.sloeber.ui.Import_Libraries_Page.createControl(Import_Libraries_Page.java:58)
	at org.eclipse.jface.wizard.WizardDialog.showStartingPage(WizardDialog.java:1218)
	at org.eclipse.jface.wizard.WizardDialog.createContents(WizardDialog.java:582)
	at org.eclipse.jface.window.Window.create(Window.java:426)
	at org.eclipse.jface.dialogs.Dialog.create(Dialog.java:1096)
	at org.eclipse.jface.window.Window.open(Window.java:783)
	at io.sloeber.ui.actions.AddLibraryAction.execute(AddLibraryAction.java:48)
	at org.eclipse.ui.internal.handlers.HandlerProxy.execute(HandlerProxy.java:291)
	at org.eclipse.ui.internal.handlers.E4HandlerProxy.execute(E4HandlerProxy.java:92)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:55)
	... 36 more
`

This is the Session Data:
`eclipse.buildId=unknown
java.version=1.8.0_131
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=de_DE
Framework arguments:  -perspective io.sloeber.application.perspective
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -perspective io.sloeber.application.perspective
`

jantje, 2018-05-02:
Are you on 4.2? I ask as there is no more code at io.sloeber.core.tools.Libraries.findAllSubFolders(Libraries.java:77) at  in the nightly

Panther2804, 2018-05-03:
Before i created the Issue.
I use the stable Version V4.2_win64.2017-11-20_02-16-52
I did a clean reinstall just to make sure, i document everything i do.
1.Open in a clean Workspace
2.Fix the Indexer Stuff
3.New Arduino Sketch
4.Open `.ino` file
5.`Add Library` works (shows intended Dialog)
6.After Playing around with what to iclude for a bit, i manged to recreate the bug:
In my original Arduino folder i had a file `README` without file type declarition e.g. `.txt`, this causes the "Add Library" function to fail.

jantje, 2018-05-03:
I can not reproduce this. I think that is so because it is fixed in https://github.com/Sloeber/arduino-eclipse-plugin/commit/1f114f571ccad9780528da7e85d3fab64bef454f#diff-15740634e0173cacba9bd960beb0edd1

closed 2018-05-07 by jantje