Sloeber/arduino-eclipse-plugin #975: Unable to compile project after sharing to git

francisreader, 2018-05-05:
Fresh install of 4.2.0.20171121011 into oxygen.3a (4.7.3a)
Unable to compile project after moving it into git.

Create new Arduino sketch with default cpp. Empty sketch Compiles correctly, even after several makes/cleans

Team->Share Project and create new project git repo. (It moves the project from workspace dir, to local git repos dir .)

Run Arduino -> Verify:
* archiver looking at previous workspace location
* workspace is in my ~/Dropbox/workspace
* git repo is in ~/git/foolocal

18:00:06 **** Build of configuration Release for project foolocal ****
make all 
Building file: ../foolocal.cpp
Starting C++ compile
"/usr/local/home/francis/eclipse/java-oxygen/eclipse//arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I"/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino" -I"/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/variants/standard" -MMD -MP -MF"foolocal.cpp.d" -MT"foolocal.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../foolocal.cpp"  -o  "foolocal.cpp.o"
Finished building: ../foolocal.cpp
 
Building file: /usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/CDC.cpp
Starting C++ compile
"/usr/local/home/francis/eclipse/java-oxygen/eclipse//arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I"/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino" -I"/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/variants/standard" -MMD -MP -MF"core/core/CDC.cpp.d" -MT"core/core/CDC.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/CDC.cpp"  -o  "core/core/CDC.cpp.o"
Finished building: /usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/CDC.cpp
 
Building file: /usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/HardwareSerial.cpp
Starting C++ compile
"/usr/local/home/francis/eclipse/java-oxygen/eclipse//arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I"/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino" -I"/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/variants/standard" -MMD -MP -MF"core/core/HardwareSerial.cpp.d" -MT"core/core/HardwareSerial.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/HardwareSerial.cpp"  -o  "core/core/HardwareSerial.cpp.o"
Finished building: /usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/HardwareSerial.cpp
 
Building file: /usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/HardwareSerial0.cpp
Starting C++ compile
"/usr/local/home/francis/eclipse/java-oxygen/eclipse//arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I"/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino" -I"/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/variants/standard" -MMD -MP -MF"core/core/HardwareSerial0.cpp.d" -MT"core/core/HardwareSerial0.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/HardwareSerial0.cpp"  -o  "core/core/HardwareSerial0.cpp.o"
Finished building: /usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/HardwareSerial0.cpp
 
Building file: /usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/HardwareSerial1.cpp
Starting C++ compile
"/usr/local/home/francis/eclipse/java-oxygen/eclipse//arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I"/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino" -I"/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/variants/standard" -MMD -MP -MF"core/core/HardwareSerial1.cpp.d" -MT"core/core/HardwareSerial1.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/HardwareSerial1.cpp"  -o  "core/core/HardwareSerial1.cpp.o"
Finished building: /usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/HardwareSerial1.cpp
 
Building file: /usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/HardwareSerial2.cpp
Starting C++ compile
"/usr/local/home/francis/eclipse/java-oxygen/eclipse//arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I"/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino" -I"/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/variants/standard" -MMD -MP -MF"core/core/HardwareSerial2.cpp.d" -MT"core/core/HardwareSerial2.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/HardwareSerial2.cpp"  -o  "core/core/HardwareSerial2.cpp.o"
Finished building: /usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/HardwareSerial2.cpp
 
Building file: /usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/HardwareSerial3.cpp
Starting C++ compile
"/usr/local/home/francis/eclipse/java-oxygen/eclipse//arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I"/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino" -I"/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/variants/standard" -MMD -MP -MF"core/core/HardwareSerial3.cpp.d" -MT"core/core/HardwareSerial3.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.21/cores/arduino/HardwareSerial3.cpp"  -o  "core/core/HardwareSerial3.cpp.o"
[... cut ...]
/usr/local/home/francis/eclipse/java-oxygen/eclipse/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ar: /usr/local/home/francis/Dropbox/workspace/foolocal/Release/arduino.ar: No such file or directory
makefile:73: recipe for target '/usr/local/home/francis/Dropbox/workspace/foolocal/Release/arduino.ar' failed
make: *** [/usr/local/home/francis/Dropbox/workspace/foolocal/Release/arduino.ar] Error 1

18:00:07 Build Finished (took 765ms)
```

jantje, 2018-05-05:
@francisreader 
Thanks for being a patron.
After moving to git you have to go to project properties->arduino and select apply&close.
Also delete the release folder.
The reason for this is that importing in git(hub) by default moves the files to another location.
Sloeber does not "see" this change so still works from the old location.
If this fixes your issue please close this issue

jantje, 2018-05-11:
Does this fix your issue? If so this issue can be closed as it is a known limitation.

francisreader, 2018-05-11:
Yes, but can you add it to your notes on how to use the plugin. NB: Found
similar issues having moved projects and now know how to resolve.

jantje, 2018-05-12:
It is the first goto thing. And indeed it is not as well documented as it should be.

closed 2020-04-21 by jantje

jonasinta, 2021-05-30:
> @francisreader
> Thanks for being a patron.
> After moving to git you have to go to project properties->arduino and select apply&close.
> Also delete the release folder.
> The reason for this is that [...]
ThankYou for this solution. Many thanks.