Sloeber/arduino-eclipse-plugin #977: Build configurations Debug and Release use very different includes in cmdline

Bascy, 2018-05-17:
I have two build configurations defined in my project, the only difference being the definition of a symob with a -DDEBUG_SENSE parameter.

Yet when comparing the commandlines used to compile one of the sources, the release config doesnt include lots of library files/folders which are needed for compiling that source.

**Building with Debug config**
> Building file: ..\CSenseAirPage.cpp'
> 'Starting C++ compile'
> "D:\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ 
> "-ID:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/include" 
> "-ID:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/lwip/include" 
> "-ID:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/libc/xtensa-lx106-elf/include" "-ID:\C-Sense2.0\source\csMainMaster/Debug/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -DF_CPU=160000000L -DLWIP_OPEN_SRC  -DARDUINO=10802 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DESP8266  
> -I"D:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266" 
> -I"D:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0-rc2\variants\nodemcu" 
> -I"D:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WiFi" 
> -I"D:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\ESP8266WiFi\src" -I"D:\C-Sense2.0\source\libraries\ic2devlib" 
> -I"D:\C-Sense2.0\source\libraries\MemoryFree" 
> -I"D:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\SoftwareSerial" 
> -I"D:\C-Sense2.0\source\libraries\Timer" 
> -I"D:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\Wire" 
> -I"D:\C-Sense2.0\source\libraries\SenseFramework" 
> -I"D:\C-Sense2.0\source\libraries\SenseFramework\src" 
> -I"D:\C-Sense2.0\source\libraries\SenseIBCFramework" 
> -I"D:\C-Sense2.0\source\libraries\SenseIBCFramework\src" 
> -I"D:\C-Sense2.0\source\libraries\SenseNextionFramework" 
> -I"D:\C-Sense2.0\source\libraries\SenseNextionFramework\src" 
> -I"D:\C-Sense2.0\source\libraries\CircularBuffer" 
> -I"D:\C-Sense2.0\source\libraries\CircularBuffer\src" 
> -I"D:\C-Sense2.0\source\libraries\MCP_CAN_lib" 
> -I"D:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\SPI" 
> -I"D:\C-Sense2.0\source\libraries\NeoNextion" -MMD -MP -MF"CSenseAirPage.cpp.d" 
> -MT"CSenseAirPage.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\CSenseAirPage.cpp"  
> -o  "CSenseAirPage.cpp.o"  -Wreorder -DDEBUG_SENSE -std=c++0x

**Building with Release config:**
> Building file: ..\CSenseAirPage.cpp'
> 'Starting C++ compile'
> "D:\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ 
> "-ID:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/include" 
> "-ID:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/lwip/include" 
> "-ID:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.0-rc2/tools/sdk/libc/xtensa-lx106-elf/include" "-ID:\C-Sense2.0\source\csMainMaster/Release/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -DF_CPU=160000000L -DLWIP_OPEN_SRC  -DARDUINO=10802 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DESP8266  
> -I"D:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266" 
> -I"D:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0-rc2\variants\nodemcu" -MMD -MP -MF"CSenseAirPage.cpp.d" 
> -MT"CSenseAirPage.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\CSenseAirPage.cpp"  
> -o  "CSenseAirPage.cpp.o"  -Wreorder	 -std=c++0x

Bascy, 2018-06-06:
Any thoughts on this? Is this a bug or a feature?

Bascy, 2018-06-06:
What have I done wrong then?

jantje, 2018-06-06:
I don't see this in my multi configuration projects and it is perfectly normal and desirable to have different include paths in different configurations.
So you may think it is a problem where it is not or you may have configured it to be that way.
For me you are requesting support and not reporting a issue.

closed 2018-06-06 by Bascy