Sloeber/arduino-eclipse-plugin #981: Unable to find a register to spill in class 'NO_REGS'

J-Rios, 2018-05-24:
Common avr-gcc compiler bug detected in v4.2 release.

_Error: unable to find a register to spill in class 'NO_REGS'_

Maybe the version of avr-gcc provided in Sloeber plugin is based on an old Arduino IDE version? This bug was detected too in Arduino IDE 1.6.X versions...

Default optimization level is **-Os**, I try with differents ones, but same error appears.

Log:
```
[Building stuffs...]
.
.
.
'Starting archiver'
"C:\Eclipse\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc-ar" rcs    C:/Users/usuario/Documents/sloeber-workspace/PROJECT_NAME/Release/arduino.ar  .\core\core\CDC.cpp.o .\core\core\HardwareSerial.cpp.o .\core\core\HardwareSerial0.cpp.o .\core\core\HardwareSerial1.cpp.o .\core\core\HardwareSerial2.cpp.o .\core\core\HardwareSerial3.cpp.o .\core\core\IPAddress.cpp.o .\core\core\PluggableUSB.cpp.o .\core\core\Print.cpp.o .\core\core\Stream.cpp.o .\core\core\Tone.cpp.o .\core\core\USBCore.cpp.o .\core\core\WInterrupts.c.o .\core\core\WMath.cpp.o .\core\core\WString.cpp.o .\core\core\abi.cpp.o .\core\core\hooks.c.o .\core\core\main.cpp.o .\core\core\new.cpp.o .\core\core\wiring.c.o .\core\core\wiring_analog.c.o .\core\core\wiring_digital.c.o .\core\core\wiring_pulse.S.o .\core\core\wiring_pulse.c.o .\core\core\wiring_shift.c.o 
'Finished building: C:\Users\usuario\Documents\sloeber-workspace\PROJECT_NAME\Release\arduino.ar'
' '
'Starting combiner'
"C:\Eclipse\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc" -Wall -Wextra -Os -Wl,--gc-sections,--relax -mmcu=atmega2560 -o "C:\Users\usuario\Documents\sloeber-workspace\PROJECT_NAME/Release/PROJECT_NAME.elf"    .\ApiRest.cpp.o .\EepromSL.cpp.o .\Globals.cpp.o .\HESystem.cpp.o .\Led.cpp.o .\Mqttcom.cpp.o .\PrLuces.cpp.o .\PrSensors.cpp.o .\Serialcmd.cpp.o .\WatchdogCTL.cpp.o .\sloeber.ino.cpp.o  .\src\pubsubclient\src\PubSubClient.cpp.o  .\src\Bridge\src\Bridge.cpp.o .\src\Bridge\src\BridgeClient.cpp.o .\src\Bridge\src\BridgeSSLClient.cpp.o .\src\Bridge\src\BridgeServer.cpp.o .\src\Bridge\src\BridgeUdp.cpp.o .\src\Bridge\src\Console.cpp.o .\src\Bridge\src\FileIO.cpp.o .\src\Bridge\src\HttpClient.cpp.o .\src\Bridge\src\Mailbox.cpp.o .\src\Bridge\src\Process.cpp.o  .\libraries\SRAM\sram.cpp.o  .\libraries\SPI\src\SPI.cpp.o  .\libraries\Ethernet\src\utility\socket.cpp.o .\libraries\Ethernet\src\utility\w5100.cpp.o  .\libraries\Ethernet\src\Dhcp.cpp.o .\libraries\Ethernet\src\Dns.cpp.o .\libraries\Ethernet\src\Ethernet.cpp.o .\libraries\Ethernet\src\EthernetClient.cpp.o .\libraries\Ethernet\src\EthernetServer.cpp.o .\libraries\Ethernet\src\EthernetUdp.cpp.o    C:/Users/usuario/Documents/sloeber-workspace/PROJECT_NAME/Release/arduino.ar  "-LC:\Users\usuario\Documents\sloeber-workspace\PROJECT_NAME/Release" -lm
C:\Eclipse\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\cores\arduino\main.cpp: In function 'main':
C:\Eclipse\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\cores\arduino\main.cpp:51:1: error: unable to find a register to spill in class 'NO_REGS'
 }
 ^
C:\Eclipse\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\cores\arduino\main.cpp:51:1: error: this is the insn:
(insn 3985 3982 3988 194 (set (mem:QI (post_dec:HI (reg/f:HI 32 __SP_L__)) [0  S1 A8])
        (subreg:QI (reg/f:HI 2817) 1)) ..\PROJECT_NAME.ino:151 1 {pushqi1}
     (expr_list:REG_ARGS_SIZE (const_int 11 [0xb])
        (nil)))
C:\Eclipse\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\cores\arduino\main.cpp:51: confused by earlier errors, bailing out
lto-wrapper: C:\Eclipse\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc returned 1 exit status
c:/eclipse/sloeber/arduinoplugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
makefile:116: recipe for target 'PROJECT_NAME.elf' failed
make: *** [PROJECT_NAME.elf] Error 1

12:59:27 Build Finished (took 35s.896ms)
```

jantje, 2018-05-24:
Sloeber uses the exact same platforms (that is boards/compilers/cofigurations°) as the arduino IDE.
The only difference is that you can use multiple platforms at the same time. 

As you can see from your output you are using the arduino platform 1.6.20. `C:\Eclipse\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20`

If -as you say- arduino platform 1.6.20 has this problem and your project uses Arduino platform 1.6.20 you will experience this problem.

To fix this I propose to just upgrade your project to a newer arduino platform. You can do so in the project properties
[image]
Select apply OK and verify your sketch. The problem should be gone now.

If there is no newer platform installed, add it to your installed platforms by selecting it in the Arduino preferences
[image]
If there is no newer platform, make sure that the local json files get updated
[image]
Turn on Update local json files and restart sloeber (the update happens at startup)

That is all there is to it.

*Sloeber does some different things with the commands in regards to warnings

J-Rios, 2018-05-24:
Thanks @jantje 

Downgrade to Arduino core 1.6.10 following the way you said, solved the issue :)

closed 2018-05-24 by J-Rios