Sloeber/arduino-eclipse-plugin #983: Eclipse indexer syntax highlighting on std::function, std::shared_ptr and lambda's

hreintke, 2018-05-27:
This is an example of the issue I see : 

[image]

On internet the generic solution for this is to  include __cplusplus = 201103L to the properties.
But that is already in the project defined.

Do you have an idea hoe to solve this ?

hreintke, 2018-05-27:
Agree that it most probably is a cdt problem.
Hoped you had seen it before and knew a solution for it.

closed 2018-05-27 by hreintke

jantje, 2018-05-27:
I know CDT is constantly working on improving so it is likely to be fixed one day.
Maybe this wil fix it: https://bugs.eclipse.org/bugs/show_bug.cgi?id=519819
Are you on the latest CDT? 9.4.3

hreintke, 2018-05-28:
yes, I am. 
The filename is `org.eclipse.cdt_9.4.3.201802261533` 
Will need to do more internet searching and/or bug reporting.

Thanks for helping.

hreintke, 2018-05-28:
Searching for solution/comparison I did a re-index of the project in sloeber.
Now all is working as expected.

Do see some small differences between C++ IDE and sloeber.
f.e. color coding of some functions, the printf is colored in C++ IDE and "plain text" in slober.

Not enough to do further investigation for now.

jantje, 2018-05-28:
Sloeber is a layer on top of CDT feeding CDT with Arduino information.
Sloeber does have its own implementation for making the makefile but further on everything is plain old CDT
Which means that from a indexer and colouring perspective everything "Should be the same". Practise however is a different story. Install history, order of loading plugin's ... so many things can explain "differences"

>Not enough to do further investigation for now.

:-)))

hreintke, 2018-05-28:
To be clear.

I did not want to say that I expected you to investigate. 
Just that I will leave it as is :-)