Sloeber/arduino-eclipse-plugin #984: esp8266 - eagle.app.v6.common.ld is not created on build - build fails

liebman, 2018-05-27:
Using the current git version of esp8266/Arduino has builds failing with the error:

> /Users/chris.l/Dropbox/git/ESPNTPServer/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot open linker script file ../ld/eagle.app.v6.common.ld: No such file or directory

and the platform.txt file has the rule:

> ## Build the app.ld linker file
> recipe.hooks.linking.prelink.1.pattern="{compiler.path}{compiler.c.cmd}" -CC -E -P {build.vtable_flags} "{runtime.platform.path}/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "{runtime.platform.path}/tools/sdk/ld/eagle.app.v6.common.ld"

I believe this is to support being able to chose the vtable memory location (flash/iram/heap).

Sloeber does not seem to be processing this rule.

jantje, 2018-05-27:
recipe.hooks.linking.prelink.1.pattern is currently not supported actually none of these "extra build commands" is currently supported.
I was thinking about adding pre and post build #927 but that was not a full solution.
I guess I'll have to take a deeper look now :-s

jantje, 2018-06-03:
Could you try to replace in platform.txt
`recipe.ar.pattern=XXXX`
with
recipe.ar.pattern=ecipe.hooks.linking.prelink.1.pattern;XXXX
After saving the file open the project properties select apply and ok and rebuild?

liebman, 2018-06-05:
That works.

jantje, 2018-06-05:
@liebman 
Great :-) Thanks for the feedback
Can you share your full platform.txt as my first take was buggy and a patron has the same issue?

liebman, 2018-06-05:
Its the stock one from the esp8266/Arduino git repo, master branch: https://github.com/esp8266/Arduino/blob/master/platform.txt  - not one I really want to have to modify to be able to compile with the latest esp8266 core.

liebman, 2018-06-06:
Diff:
```
diff --git a/platform.txt b/platform.txt
index 429cea6e..520b549c 100644
--- a/platform.txt
+++ b/platform.txt
@@ -91,7 +91,7 @@ recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}" {compiler.cpreprocessor
 recipe.S.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.cpreprocessor.flags} {compiler.S.flags} -DF_CPU={build.f_cpu} {build.lwip_flags} {build.debug_port} {build.debug_level} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} -DARDUINO_BOARD="{build.board}" {build.led} {compiler.c.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"
 
 ## Create archives
-recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/arduino.ar" "{object_file}"
+recipe.ar.pattern={recipe.hooks.linking.prelink.1.pattern};"{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/arduino.ar" "{object_file}"
 
 ## Combine gc-sections, archives, and objects
 recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" -Wl,-Map "-Wl,{build.path}/{build.project_name}.map" {compiler.c.elf.flags} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" -Wl,--start-group {object_files} "{build.path}/arduino.ar" {compiler.c.elf.libs} -Wl,--end-group  "-L{build.path}"
```

full file:
```

# ESP8266 platform
# ------------------------------

# For more info:
# https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification

name=ESP8266 Modules
version=2.5.0-dev

runtime.tools.xtensa-lx106-elf-gcc.path={runtime.platform.path}/tools/xtensa-lx106-elf
runtime.tools.esptool.path={runtime.platform.path}/tools/esptool

compiler.warning_flags=-w
compiler.warning_flags.none=-w
compiler.warning_flags.default=
compiler.warning_flags.more=-Wall
compiler.warning_flags.all=-Wall -Wextra

build.lwip_lib=-llwip_gcc
build.lwip_include=lwip/include
build.lwip_flags=-DLWIP_OPEN_SRC

build.vtable_flags=-DVTABLES_IN_FLASH

build.float=-u _printf_float -u _scanf_float
build.led=

compiler.path={runtime.tools.xtensa-lx106-elf-gcc.path}/bin/
compiler.sdk.path={runtime.platform.path}/tools/sdk
compiler.libc.path={runtime.platform.path}/tools/sdk/libc/xtensa-lx106-elf
compiler.cpreprocessor.flags=-D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I{compiler.sdk.path}/include" "-I{compiler.sdk.path}/{build.lwip_include}" "-I{compiler.libc.path}/include" "-I{build.path}/core"

compiler.c.cmd=xtensa-lx106-elf-gcc
compiler.c.flags=-c {compiler.warning_flags} -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -ffunction-sections -fdata-sections

compiler.S.cmd=xtensa-lx106-elf-gcc
compiler.S.flags=-c -g -x assembler-with-cpp -MMD -mlongcalls

compiler.c.elf.flags=-g {compiler.warning_flags} -Os -nostdlib -Wl,--no-check-sections -u app_entry {build.float} -Wl,-static "-L{compiler.sdk.path}/lib" "-L{compiler.sdk.path}/ld" "-L{compiler.libc.path}/lib" "-T{build.flash_ld}" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read

compiler.c.elf.cmd=xtensa-lx106-elf-gcc
compiler.c.elf.libs=-lhal -lphy -lpp -lnet80211 {build.lwip_lib} -lwpa -lcrypto -lmain -lwps -lbearssl -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc

compiler.cpp.cmd=xtensa-lx106-elf-g++
compiler.cpp.flags=-c {compiler.warning_flags} -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections

compiler.as.cmd=xtensa-lx106-elf-as

compiler.ar.cmd=xtensa-lx106-elf-ar
compiler.ar.flags=cru
[... 58 lines cut ...]

## Compute size
recipe.size.pattern="{compiler.path}{compiler.size.cmd}" -A "{build.path}/{build.project_name}.elf"
recipe.size.regex=^(?:\.irom0\.text|\.text|\.data|\.rodata|)\s+([0-9]+).*
recipe.size.regex.data=^(?:\.data|\.rodata|\.bss)\s+([0-9]+).*
#recipe.size.regex.eeprom=^(?:\.eeprom)\s+([0-9]+).*

# ------------------------------

tools.esptool.cmd=esptool
tools.esptool.cmd.windows=esptool.exe
tools.esptool.path={runtime.platform.path}/tools/esptool
tools.esptool.network_cmd=python
tools.esptool.network_cmd.windows=python.exe

tools.esptool.upload.protocol=esp
tools.esptool.upload.params.verbose=-vv
tools.esptool.upload.params.quiet=
tools.esptool.upload.pattern="{path}/{cmd}" {upload.verbose} -cd {upload.resetmethod} -cb {upload.speed} -cp "{serial.port}" {upload.erase_cmd} -ca 0x00000 -cf "{build.path}/{build.project_name}.bin"
tools.esptool.upload.network_pattern="{network_cmd}" "{runtime.platform.path}/tools/espota.py" -i "{serial.port}" -p "{network.port}" "--auth={network.password}" -f "{build.path}/{build.project_name}.bin"

tools.mkspiffs.cmd=mkspiffs
tools.mkspiffs.cmd.windows=mkspiffs.exe
tools.mkspiffs.path={runtime.platform.path}/tools/mkspiffs
```

jantje, 2018-06-14:
Seems the workaround only works in Linux because linus supports the ; to have multiple commands on one line.
I had expected make to handle this (as it did in other cases) but apparently it doesn't

jantje, 2018-06-14:
The workaround should actually be done before the combiner and not the archiver
For windows one should use & and not ; to seperate the commands
So
recipe.c.combine.pattern=XXXX
on linux/mac should be replaced with
recipe.c.combine.pattern={recipe.hooks.linking.prelink.1.pattern};XXXX
on windows should be replaced with
recipe.c.combine.pattern={recipe.hooks.linking.prelink.1.pattern} & XXXX

liebman, 2018-06-15:
I'm on a Mac - I can test there. (mostly similar)

jantje, 2018-06-15:
also good :-)

liebman, 2018-06-16:
Works nicely on a Mac!

closed 2018-06-16 by liebman

jantje, 2018-06-16:
great thanks.

michael-uman, 2019-12-05:
This issue has not been resolved. I am using the latest build of Eclipse and the current ESP8266 board installed yet I still get this error. I find it hard to believe this is still an issue after all this time. I have tried making the changes which are recommended in this thread but this does not work.

Anyone out there have a clue what it takes to be able to build 8266 code in Arduino plugin for Eclipse? I can make ESP32, Arduino Uno & Mega firmware but ESP8266 has always failed when it tried to find this linker script.

KareemMohamed1, 2020-01-22:
i faced the same problem . and then i have tried to hack the platform.txt with the solution provided by @jantje  but i still have the same problem, it didnt work with me , 
Can anyone give a hand on thins, thanks in advance