Sloeber/arduino-eclipse-plugin #991: Building workspace causes out of memory error  at 62%

Bascy, 2018-06-13:
I'm recently running into a stalling Eclipse, eventually resulting in a Out of Memory Error, when I open my Eclipse workspace. Building Workspace is at 62%, "Convert INO to CPP files"
Sometimes it helps to revert to an earlier version of the .cproject or .project file but I havent been able to make this reproducible or to reproduce what to do to fix it.

My suspicions are towards a recursing problem causing a stackoverflow causing the Out of memory error. What log files can I inspect to get more info on what is actually going on?

Bascy, 2018-06-13:
Found some exceptions in .metadata\.log

!ENTRY org.eclipse.cdt.core 4 0 2018-06-14 01:17:42.308
!MESSAGE Can't load preferences from file D:/C-Sense2.0/source/csWaterflowCANNode/.settings/language.settings.xml
!STACK 1
org.eclipse.core.runtime.CoreException: Internal error while trying to load XML document
	at org.eclipse.cdt.internal.core.XmlUtil.loadXml(XmlUtil.java:264)
	at org.eclipse.cdt.internal.core.language.settings.providers.LanguageSettingsProvidersSerializer.loadLanguageSettings(LanguageSettingsProvidersSerializer.java:1145)
	at org.eclipse.cdt.internal.core.settings.model.xml.XmlProjectDescriptionStorage.loadProjectDescription(XmlProjectDescriptionStorage.java:499)
	at org.eclipse.cdt.internal.core.settings.model.xml.XmlProjectDescriptionStorage.getProjectDescription(XmlProjectDescriptionStorage.java:237)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescriptionInternal(CProjectDescriptionManager.java:426)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescription(CProjectDescriptionManager.java:408)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescription(CProjectDescriptionManager.java:402)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescription(CProjectDescriptionManager.java:395)
	at org.eclipse.cdt.internal.core.settings.model.CProjectDescriptionManager.getProjectDescription(CProjectDescriptionManager.java:630)
	at org.eclipse.cdt.core.model.CoreModel.getProjectDescription(CoreModel.java:1372)
	at io.sloeber.ui.ExplorerLabelDecorator.decorateText(ExplorerLabelDecorator.java:49)
	at org.eclipse.ui.internal.decorators.FullDecoratorDefinition.decorateText(FullDecoratorDefinition.java:133)
	at org.eclipse.ui.internal.decorators.FullTextDecoratorRunnable.run(FullTextDecoratorRunnable.java:24)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.ui.internal.decorators.DecoratorManager.safeDecorateText(DecoratorManager.java:497)
	at org.eclipse.ui.internal.decorators.DecoratorManager.decorateText(DecoratorManager.java:453)
	at org.eclipse.ui.internal.decorators.DecoratorManager$ManagedWorkbenchLabelDecorator.decorateText(DecoratorManager.java:151)
	at org.eclipse.jface.viewers.DecoratingStyledCellLabelProvider.getStyledText(DecoratingStyledCellLabelProvider.java:202)
	at org.eclipse.jface.viewers.DelegatingStyledCellLabelProvider.update(DelegatingStyledCellLabelProvider.java:106)
	at org.eclipse.jface.viewers.DecoratingStyledCellLabelProvider.update(DecoratingStyledCellLabelProvider.java:131)
	at org.eclipse.jface.viewers.ViewerColumn.refresh(ViewerColumn.java:141)
	at org.eclipse.jface.viewers.AbstractTreeViewer.doUpdateItem(AbstractTreeViewer.java:946)
	at org.eclipse.jface.viewers.AbstractTreeViewer$UpdateItemSafeRunnable.run(AbstractTreeViewer.java:117)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.ui.internal.JFaceUtil.lambda$0(JFaceUtil.java:44)
	at org.eclipse.jface.util.SafeRunnable.run(SafeRunnable.java:173)
	at org.eclipse.jface.viewers.AbstractTreeViewer.doUpdateItem(AbstractTreeViewer.java:1025)
	at org.eclipse.jface.viewers.StructuredViewer$UpdateItemSafeRunnable.run(StructuredViewer.java:475)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.ui.internal.JFaceUtil.lambda$0(JFaceUtil.java:44)
	at org.eclipse.jface.util.SafeRunnable.run(SafeRunnable.java:173)
	at org.eclipse.jface.viewers.StructuredViewer.updateItem(StructuredViewer.java:2159)
	at org.eclipse.jface.viewers.AbstractTreeViewer.createTreeItem(AbstractTreeViewer.java:840)
	at org.eclipse.jface.viewers.AbstractTreeViewer.lambda$0(AbstractTreeViewer.java:817)
	at org.eclipse.swt.custom.BusyIndicator.showWhile(BusyIndicator.java:70)
	at org.eclipse.jface.viewers.AbstractTreeViewer.createChildren(AbstractTreeViewer.java:792)
	at org.eclipse.jface.viewers.TreeViewer.createChildren(TreeViewer.java:595)
	at org.eclipse.jface.viewers.AbstractTreeViewer.createChildren(AbstractTreeViewer.java:763)
	at org.eclipse.jface.viewers.AbstractTreeViewer.internalInitializeTree(AbstractTreeViewer.java:1527)
	at org.eclipse.jface.viewers.TreeViewer.internalInitializeTree(TreeViewer.java:771)
	at org.eclipse.jface.viewers.AbstractTreeViewer.lambda$1(AbstractTreeViewer.java:1512)
	at org.eclipse.jface.viewers.StructuredViewer.preservingSelection(StructuredViewer.java:1447)
	at org.eclipse.jface.viewers.TreeViewer.preservingSelection(TreeViewer.java:354)
	at org.eclipse.jface.viewers.StructuredViewer.preservingSelection(StructuredViewer.java:1408)
	at org.eclipse.jface.viewers.AbstractTreeViewer.inputChanged(AbstractTreeViewer.java:1506)
	at org.eclipse.jface.viewers.ContentViewer.setInput(ContentViewer.java:286)
	at org.eclipse.jface.viewers.StructuredViewer.setInput(StructuredViewer.java:1685)
	at org.eclipse.ui.navigator.CommonNavigator.createPartControl(CommonNavigator.java:206)
	at org.eclipse.ui.navigator.resources.ProjectExplorer.createPartControl(ProjectExplorer.java:88)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.createPartControl(CompatibilityPart.java:151)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityView.createPartControl(CompatibilityView.java:147)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.create(CompatibilityPart.java:355)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[... cut ...]
	at org.eclipse.cdt.internal.core.model.TranslationUnit.getAST(TranslationUnit.java:880)
	at org.eclipse.cdt.internal.core.model.TranslationUnit.getAST(TranslationUnit.java:838)
	at io.sloeber.core.tools.PdePreprocessor.extendHeaderForFile(PdePreprocessor.java:156)
	at io.sloeber.core.tools.PdePreprocessor.processProject(PdePreprocessor.java:103)
	at io.sloeber.core.builder.inoToCpp$SampleDeltaVisitor.visit(inoToCpp.java:30)
	at org.eclipse.core.internal.events.ResourceDelta.accept(ResourceDelta.java:64)

jantje, 2018-06-13:
It looks as if the label decorator hits in when converting ino to cpp files which causes CDT to be confused but I have no clue about what it is confused).
Are you on the latest nightly? I remember doing changes in the area of decorators coping will null pointers.
You could turn off the decorator and see if that helps.

Bascy, 2018-06-13:
Running on 4.2
How do I turn off decorators?

jantje, 2018-06-14:
In the preferences. I can't tell you exactly as I'm running in dutch right now. but look for decorator in the preferences

Bascy, 2018-06-14:
Should I disable all decorators or just the "embedded development" one?

jantje, 2018-06-14:
only the sloeber one that shows you the board and port

jantje, 2018-06-18:
I have no clue what is going on at your system.
Is this a big project?
Are there many projects in your workspace?
Which java version do you use?
Are you using the sloeber product or the plugin?
What are the -Xms256m and -Xmx1024m settings in your eclipse.ini file

Bascy, 2018-06-18:
Yes it is a big project
The workspace contains for projects, all sharing the use of a number of big libraries
Jave version used, is the one provided with Sloeber  Eclipse,  java version "1.8.0_131"
I'm using the sloeber product 4.2

And the sloeber-ide.ini contains the following:

> -startup
> plugins/org.eclipse.equinox.launcher_1.4.0.v20161219-1356.jar
> -perspective
> io.sloeber.application.perspective
> --launcher.defaultAction
> openFile
> --launcher.appendVmargs
> --launcher.library
> plugins/org.eclipse.equinox.launcher.win32.win32.x86_64_1.1.551.v20171108-1834
> -vmargs
> -Dosgi.requiredJavaVersion=1.8
> -Dorg.eclipse.update.reconcile=false
> -Xms40m
> -Xmx512m
> --add-modules=ALL-SYSTEM
> 
>

jantje, 2018-06-18:
eclipse now sets
-Xms256m
-Xmx1024m

I would give that a try

Bascy, 2018-06-19:
thanks for the tip, will try that tonight

jantje, 2018-06-19:
I think you are waiting for the indexer.
Do you really need the ino file format?

Bascy, 2018-06-19:
No, if we can avoid it, please tell us how

jantje, 2018-06-19:
rename the .ino to .cpp
you will need to define methods before usage but I guess you are used to c++ programming

jantje, 2018-07-15:
Initial problem is solved so closing

closed 2018-07-15 by jantje