Sloeber/arduino-eclipse-plugin #992: Error on make clean

winginitau, 2018-06-14:
G'day,

Have been ignoring this for a while (not critical). It started happening right after a nightly build update some time back (4-5 weeks?). Have been meaning to report it but only just getting around to it. It happens when I make clean any sloeber project.

Log:

22:11:54 **** Clean-only build of configuration Release for project ff_arduino ****
make clean 
rm  ./libraries/Wire/src/utility/twi.c.d  ./core/core/WInterrupts.c.d ./core/core/hooks.c.d 
...
...
...
./core/core/USBCore.cpp.d ./core/core/WMath.cpp.d ./core/core/WString.cpp.d ./core/core/abi.cpp.d ./core/core/main.cpp.d ./core/core/new.cpp.d 
rm: cannot remove './ff_arduino.ino.d': No such file or directory
rm: cannot remove 'ff_arduino.elf': No such file or directory
makefile:111: recipe for target 'clean' failed
make: [clean] Error 1 (ignored)

jantje, 2018-06-14:
have you done the update procedure of the project?
http://eclipse.baeyens.it/how_to.shtml#/n

winginitau, 2018-06-14:
I've done this bit:
Party success
1. drink a beer
2. Become a patron of jantje

The rest is hard. I have 17 projects in the workspace. (including PyDev projects too). At least 4 of them have shared resources with lots of exclude-from-build filters applied. To rebuild the workspace is a multi-hour process. 

I might opt out of the nightly for now and continue to ignore the make clean error.

jantje, 2018-06-14:
You have a big workspace :-)
As you clearly understand the clean error is not a real problem.

The ff_arduino.elf is probably a bug that also nags me in relation to creating debug launch configurations. Does your project build a .elf or is it.bin or anything else?

ff_arduino.ino.d is a file that should never be generated (as .ino files are compiled as includes of sloeber.ino.cpp) but it should also not be in the clean.

>The rest is hard. I have 17 projects in the workspace. (including PyDev projects too). At least 4 of them have shared resources with lots of exclude-from-build filters applied. To rebuild the workspace is a multi-hour process.

There is virtually no real extra time to build a project after doing any of these 2 
1) Project properties->arduino->apply close and deleting the release folder 
2) clean of the project.
So maybe the explanation is wrong/confusing but what sometimes needs to be done after a update is 1

Off course this only needs to be done for sloeber projects and this should not impact shared resources nor exclude from builds. If so that is an alarming bug.

>I've done this bit:
>Party success
>
>   drink a beer
>    Become a patron of jantje

Thanks for making me smile and for being a patron :-)

jantje, 2018-06-15:
Apparently the .ino files are now in the makefile.
This needs to be fixed

closed 2018-06-16 by jantje

winginitau, 2018-06-16:
Fixed. Thanks!

Yes, understood now re: Project properties->arduino->apply close and deleting the release folder. 

That didn't work because of the .ino, now fixed. 

I think my toolchain builds the .elf even for release. And I do use it from time to time to look at object memory usage. Not a problem though.