Sloeber/arduino-eclipse-plugin #993: Combiner problem on new build configurations

mostwright, 2018-06-21:
I have an issue where any new build configurations I make do not want to build successfully due to static path names from old build configurations persisting in the variables that auto generate the make files for the new configuration. It seems like a different flavor of this issue [294](https://github.com/Sloeber/arduino-eclipse-plugin/issues/294)

To reproduce:

Make a new project based on the blink example; by default it has one build configuration “Release”. It compiles and uploads fine.
Make a new configuration based on “Release”. It fails to build with the following error:

```
'Starting combiner'
"C:\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc" -Wall -Wextra -Os -Wl,--gc-sections,--relax -mmcu=atmega2560 -o "C:/sloeber-workspace/laserdevice/Release/laserdevice.elf"    .\sloeber.ino.cpp.o    C:/sloeber-workspace/laserdevice/smallTempControllerV0.1.1/arduino.ar  "-LC:/sloeber-workspace/laserdevice/Release" -lm
'Finished building: laserdevice.elf'
' '
'Create eeprom image'
"C:\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0  "C:\sloeber-workspace\laserdevice/smallTempControllerV0.1.1/laserdevice.elf" "C:\sloeber-workspace\laserdevice/smallTempControllerV0.1.1/laserdevice.eep"
C:\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-objcopy: 'C:\sloeber-workspace\laserdevice/smallTempControllerV0.1.1/laserdevice.elf': No such file
makefile:85: recipe for target 'laserdevice.hex' failed
make: *** [laserdevice.hex] Error 1
```

Looking at .settings/org.eclipse.cdt.core.prefs, there are a number of variables related to the new configuration that still have “Release” hardcoded into their path, which seems to be what caused the build to fail (ex: A.RECIPE.C.COMBINE.PATTERN.1, A.RECIPE.OBJCOPY.EEP.PATTERN.1, etc). Doing a find/replace all for “Release”->”${ConfigName}” appears to solve this issue, and making additional new configurations based on the modified one also seems to work fine.

I’m not sure if this is Sloeber specific, Eclipse specific, or me being dumb specific. Sloeber v4.2_win64

jantje, 2018-06-22:
I'm using multiple configurations quite often. It works fine .... however not all ways of changing the configuration work properly.
A build all configurations for instance will not work
Also project properties->something->configurations->set active will not work
What works for me is
Right click the project->build configurations->set active
In CDT perspective select the dropdown next to the build icon and select the configuration

jantje, 2018-07-15:
Closing as to long open without input

closed 2018-07-15 by jantje