Sloeber/arduino-eclipse-plugin #994: LocalDebug Invalid Arguments Issue

human890209, 2018-06-25:
HI,
I installed 64bit stable version of Sloeber 4.2. And the Arduino plugin was installed automatically (1.6.21). I created a Blink example sketch and verified it then uploaded it to my Arduino UNO. Everything worked. No invalid arguments issue.
Then I installed the Local Debug plugin which is under Perference\Arduino\Platform and Boards\Jantje\Arduino avr Boards(local debug)\1.0.2
After the setup completed. I follow this Youtube Video [https://www.youtube.com/watch?v=ZN-rLSyhBN8] to local debug a BlinkWithoutDelay example.
And set to [Release] Build Configuration(1.6.21), everything worked.
But when changed the Build Configuration to [localdebug].
Invalid arguments were reported by indexer.
And the debug is terminated.
[image]
Then I added [#include "Arduino.h"].
Some invalid Arguments went away. But still get 2 left. pinMode() and digitalWrite() were not recognized.
[image]

It's my first time to use Sloeber, Hope somebody could help me. Thanks.

jantje, 2018-06-25:
check http://eclipse.baeyens.it/faq.shtml#/troubleshooting
section "Eclipse tells me there are bugs in my code but it compiles ok."

human890209, 2018-06-27:
I checked troubleshooting, cause I searched similar issues about Invalid Arguments before I post here.
Actually, I can't find the "windows->preferences->C/C++->indexer->Files to index upfront . " in 4.2.
And I did #include "Arduino.h" then some errors go away (value error), but some still remain (function error).
[image]

jantje, 2018-06-27:
What is the console output?

human890209, 2018-07-01:
Sorry for my late reply.

[image]
[image]
[image]
[image]
[image]

human890209, 2018-07-01:
Old Problems
[image]

And #include "Arduino.h" still 2 function error unresolved

[image]
[image]

jantje, 2018-07-01:
As far as I can see the console states the build is fine

human890209, 2018-07-03:
Why the debug is terminated? How to use the local debug?

jantje, 2018-07-03:
This is "advanced usage". See this video for more info https://youtu.be/szRrvJPvQz4

jantje, 2018-07-15:
I guess the video fixed this

closed 2018-07-15 by jantje