Sloeber/arduino-eclipse-plugin #995: Can't Upload to Adafruit Feather M0

gfvalvo, 2018-06-27:
Uploading to my Adafruit Feather M0 works fine in the Arduino IDE. With Sloeber, not so much.

I have "https://adafruit.github.io/arduino-board-index/package_adafruit_index.json" included in "Third party index URLs" and the latest packages selected under "Platforms and Boards".

The code compiles fine, but pukes on upload attempt. Displayed in Console:

Starting upload
using arduino loader
Forcing reset using 1200bps open/close on portCOM24
PORTS { COM1, COM3, COM13, COM24,} / { config error:, see https://github.com/jantje/arduino-eclipse-plugin/issues/252,} => { config error:, see https://github.com/jantje/arduino-eclipse-plugin/issues/252,}
Comport reset took250ms
Using comportconfig error: from now onwards
Ending reset

Launching/bossac --port=config error: -U true -i -e -w -v D:\SloeberWorkspace\testSleep/Release/testSleep.bin -R 
Output:
failed to upload

Could be something with the complex dual port USB arrangement of this chip. Ardunio has no problem with it, but I really need to use Sloeber.

Thanks.

jantje, 2018-06-27:
Have you looked at #252 ?

gfvalvo, 2018-06-27:
I did. But, I'm not sure what part is applicable. Nor am I interested in reading a novel. All I want to do is upload to the Feather M0. Is that possible? If so, are there simple instructions for doing so?

gfvalvo, 2018-06-27:
So, is there fix for the Sloeber bug preventing upload to this board?

First time it's happened for me. Uploads have been working fine with Uno, Teensy 3.2, and ESP8266 Feather Huzzah.

gfvalvo, 2018-06-30:
OK, there appears to be a workaround -- manually run the version of bossac installed by Arduino IDE and point it to the .bin file generated by the build Sloeber:

D:\arduino-1.8.5\portable\packages\arduino\tools\bossac\1.7.0/bossac -i -d --port=COM23 -U true -i -e -w -v D:\SloeberWorkspace\testSleep/Release/testSleep.bin -R

You can create a .bat file to do this.

closed 2018-07-15 by jantje

jantje, 2018-11-09:
I think this is actually the same thing as #1035

jantje, 2018-11-24:
This is actually a problem in the json file provided by adafruit

Workaroud:
Install "Arduino SAMD Boards (32-bits ARM Cortex-M0+)" version "1.6.7". (which spoecifies bossac 1.7 as tool)
For all affected existing projects open project properties-Arduino->apply and ok
You can safely deinstall "Arduino SAMD Boards (32-bits ARM Cortex-M0+)" as this will not remove the installed tools.

jantje, 2018-11-24:
Eventhough this is not a Sloeber limitation I mark it as such because this makes it more visible as a problem