Sloeber/arduino-eclipse-plugin #996: Sloeber-ide.exe contains virus!

hartoyo123rinan, 2018-06-30:
[image]
Kaspersky says there is a Trojan.Win32.Generic virus in sloeber-ide.exe contains virus!

Do you have a clean version?

Thanks!

jantje, 2018-06-30:
I think the download packages on the github server are clean. So should the ones be on the update sites.
Is this a new install? 
Is this the nightly?
Where did you get it?
 
I'm pretty sure there is no virus in the Sloeber files on the download servers. But I can not guarantee it, because anyone who does doesn't know what he is talking about.
There are however no binaries in the github repository so if you build Sloeber from code you can be (relatively) sure there will not be additional viruses added to your system.
Relatively sure, because to build sloeber from the repositories you need git, maven, compilers, java ...

jantje, 2018-07-15:
Do you have the input requested? More data? Is this a false positive?

jantje, 2018-08-24:
Closing due to no response from OP

closed 2018-08-24 by jantje