Sloeber/arduino-eclipse-plugin #998: No default board settings

Crazor, 2018-07-03:
On creating a new sketch, on the screen where I select the platform folder, the wizard is missing the defaults for the board settings in the lower section. Most boards only have one or two settings, but e.g. the Espressif parts have a bunch of them, like debug level, flash fequency, flash mode etc.
When I choose a board in the Arduino IDE, those settings all have default values. Sloeber has them all empty, so I have to cross-reference with the Arduino IDE.

jantje, 2018-07-03:
Sloeber has the last used but no defaults. 
The https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification does not specify how to specify the "default" option which means Arduino IDE probably takes the first one.
Which means sloeber can not guarantee the same "default" selection.

Crazor, 2018-07-05:
Interesting, I haven't noticed that. Usually, boards come up with the correct settings in the Arduino IDE, apart from the port of course. I have never thought about this being "just luck", but instead I felt like those were some kind of "factory defaults", like flash size for different boards with Espressif chips.

closed 2018-07-05 by Crazor