Sloeber/arduino-eclipse-plugin #1000: Sloeber's Auto Format Sometimes Makes a Mess of Things

gfvalvo, 2018-07-19:
So, for the most part, Sloeber's auto format function does a pretty good job. But, for things like specifying a multi-dimensional array over multiple lines it produces pretty poor results.

For example, here's what the code looks like after formatting with Arduino IDE's.
```
uint8_t byteArray[][4] =
{ {0x12, 0x4D, 0x1F, 0xFF},
  {0x9B, 0xDE, 0xCA, 0xAC},
  {0xAA, 0xCC, 0x83, 0x22},
  {0x34, 0x71, 0x9F, 0xFF}
};
```

Beautiful and easy to read right? Here's Sloeber's opinion of what it should look like:
```
uint8_t byteArray[][4] = { { 0x12, 0x4D, 0x1F, 0xFF },
		{ 0x9B, 0xDE, 0xCA, 0xAC }, { 0xAA, 0xCC, 0x83, 0x22 }, { 0x34, 0x71,
				0x9F, 0xFF } };
```
It's a mess, and it only gets worse with bigger arrays.
Any way to fix this?

Thanks.

jantje, 2018-07-19:
Thanks for finding the solution yourself :-)

closed 2018-07-19 by jantje

wimjongman, 2018-07-20:
Yay! Issue #1000!