Sloeber/arduino-eclipse-plugin #1001: Changing variable name causes crash

gunstr, 2018-07-20:
Hi,

My first post in this forum, but I hope to get some advice on a problem where I have got completely stuck.

I'm working on a project for an Industruino with a 32bit D21G top board. I came quite long and everything worked fine until suddenly the Industruino started to crash when I added more code - it hangs completely so also the USB port disappear from the PC. With a hard reset and uploading a "hello world" sketch I can start over.

My first though was that I had done wrong in the code, so I removed parts and changed structure, but after a while same thing happened. Then I though something went wrong during upload due to size of project, however it's not very big, uses ~10% of a available flash and about the same for RAM. I then found out that by adding a global bool with no other use than changing the value in the main loop I could force the crash. And even more weird, if I change the name of that variable I could toggle between code that crashes or not crashes.

I then started to remove functions and libraries from the code, and strangely enough the behavior with changing name of that dummy variable remains. I have now removed almost all code, only a few lines of code in the main sketch and a separate class with another nested class remains. More or less all functions are gone and the code does not do anything meaningful. I have not found anything more I can remove - if I do the impact on changing name of the dummy variable is gone and I see no crash.

If it's something wring with my code I can't find it, and I'm not very experienced in how the compiler works so I do not know how to analyze on a lower level.

I have created two completely new projects and imported only the three source files, with different variable name with same result - one is crashing, the other is not. Attaching both as zip files if that could help (I hope that is OK?).

[18033_Indio_crash.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/2213275/18033_Indio_crash.zip)

[18033_Indio_no_crash.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/2213280/18033_Indio_no_crash.zip)

18033_01_Indio_crash: Variable name "testVariableNewName"
18033_01_Indio_no_crash: Variable name "testVariable"

Any suggestion to what I could do is appreciated.

Rgds
Gunnar

jantje, 2018-07-20:
I have seen these things in my career before. If it doen't make sense at all it is likely a compiler bug.
With doesn't make sense I mean: it is fixed when I
-Add a variable
-Add a blank line
-Rename a method/file

I see you have done the legwork so I assume you have
-deleted the output folder (normally release)
-Check whether you have the same behaviour in the arduino ide

If that doesn't help you, IMHO contacting the board vendor is your best move.

gunstr, 2018-07-20:
Thanks,
Yes, I have deleted the output folder, and even started with fresh projects importing the source only.
But behavior in standard arduino ide is not fully consistent, I have tested several versions of the code during my "scale down" of the project and have seen same crash. However, I checked the last very slimmed code and then I do not see it.
Just for my knowledge - if nested classes are not allowed the compiler should report an error right? And if it's not supported and there is a compiler bug, is there any way for me to check what is actually supported or is that something only the board vendor can answer?
I will try to take away the nested class but also if it helps I want to make sure that was the issue.

jantje, 2018-07-20:
>Just for my knowledge - if nested classes are not allowed the compiler should report an error right?

Yes, the compiler should report a error. But live is not easy and should is not the same as does. The cases where it "should but doesn't", should be called bugs. 
There are some minor differences between the compilation commands with arduino IDE and sloeber. Having the compile output (the stuff in the console) of sloeber and arduino ide will help the gcc people  to find the root cause. The main difference is related to the building of the archive where arduino ide runs a command  for each object file to add to the library where sloeber adds all files in one go.

gunstr, 2018-07-20:
Moved the nested class to be a separate class, but it still behaves the same. Will take your advice that it's likely a compiler issue as it does not make sense at all - and check with the vendor.

gunstr, 2018-07-24:
Problem most likely resolved. After advice from the Industruino team I have scrutinized my code again focusing on memory leaks and out of bound memory access, and finally I found the error. By mistake an array index could be incremented outside the boundary. Obvious when you see it and it has been there in front of my eyes all the time, but when you get stuck it helps a lot to get a push in the right direction from someone else.

closed 2018-07-24 by gunstr

jantje, 2018-07-24:
I'm really bad at array boundaries. I mostly take the safe root of assigning some extra rows.
Not a embedded thing, I know.