Sloeber/arduino-eclipse-plugin #1002: When process "new" with STM32F103C8 "Bluepill" freeze.

human890209, 2018-07-30:
Hi,
I tried using STM32F103C8 "Bluepill" with today. Cause it is supported by Arduino IDE. 
Everything is here. https://github.com/rogerclarkmelbourne/Arduino_STM32
I tested the Blink example both with Arduino IDE and Sloeber, they both worked!
Then I continue to do more tests with Sloeber, cause it's better for me:>
Then I got this bug with "new". Here I paste the simple bug reproduce sketch code.
```
class MYCLASS
{
public:
	MYCLASS(){content = 99;};
	~MYCLASS(){};
	byte content;
};

MYCLASS mc;

MYCLASS* mcp;

void setup()
{
Serial.begin(115200);
pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
	Serial.println("loop");
	digitalWrite(LED_BUILTIN, HIGH);
	delay(500);
	digitalWrite(LED_BUILTIN, LOW);
	delay(500);
	Serial.println(mc.content);
	mcp = new MYCLASS();
	Serial.println((*mcp).content + 1);
	delete(mcp);
}
```

SerialMonitor(Sloeber):
```
Connect to serial portCOM3 at115200
99
```

I tried with the same code in Arduino IDE. Everything works fine! (Took me much time to find out it's not a pure hardware bug...Asked around on the STM32 Arduino Github Issues and Forum, cause Arduino IDE works fine, I came here to ask a way out.)

This is my project setting of Sloeber
[image]
Simply use the same default setting with Arduino IDE.

I'm sure "new" is the problem. Cause ` mcp = &mc;` works fine with Sloeber.

Hope you could help me out with the "new"+"STM32 Bluepill"+"Sloeber" bug.
It will be awesome if STM32 Bluepills can be used with Sloeber. When Flash and SRAM get larger Sloeber is in need.

human890209, 2018-07-30:
I found a warning in the compiling log. Maybe this could do some help.
```
'Starting combiner'
"C:\sloeber\/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -Os -Wl,--gc-sections -mcpu=cortex-m3 "-TD:\Arduino\Software\Sloeber\MyHardware\Arduino_STM32\STM32F1\variants\generic_stm32f103c/ld/jtag_c8.ld" "-Wl,-Map,C:\Users\human\Documents\sloeber-workspace\STM32F103C/Release/STM32F103C.map" "-LD:\Arduino\Software\Sloeber\MyHardware\Arduino_STM32\STM32F1\variants\generic_stm32f103c/ld" -o "C:\Users\human\Documents\sloeber-workspace\STM32F103C/Release/STM32F103C.elf" "-LC:\Users\human\Documents\sloeber-workspace\STM32F103C/Release" -lm -lgcc -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group    .\sloeber.ino.cpp.o    C:/Users/human/Documents/sloeber-workspace/STM32F103C/Release/arduino.ar  -Wl,--end-group
**c:/sloeber/arduinoplugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7-m\libnosys.a(sbrk.o): In function `_sbrk':
sbrk.c:(.text._sbrk+0x30): warning: undefined reference to `end'**
'Finished building: STM32F103C.elf'
```

human890209, 2018-07-31:
Here is the solution:
http://www.stm32duino.com/viewtopic.php?f=3&t=3941&p=47745#p47745

closed 2018-07-31 by human890209

jantje, 2018-07-31:
Great you fixed this yourself.

RickKimball, 2018-08-03:
The solution suggested above was a work around. The real problem has actually been fixed by #1005