Sloeber/arduino-eclipse-plugin #1003: How can I change the include folder for building debug project with Jantje

Super169, 2018-07-31:
I just try to follow the tutorial here https://www.youtube.com/watch?v=WFzVzgEA8Fo for ESP8266 debugging.

It can build the project with release configuration for ESP8266 2.4.1, but when I attempt to build with the debug configuration for Jantje, it will return fatal error: c_types.h: No such file or directory.

> 01:02:17 **** Incremental Build of configuration Release for project MyESP8266 ****
> "C:\\Util\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
> 'Building file: ..\sloeber.ino.cpp'
> 'Starting C++ compile'
> "C:\Util\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Util\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-IC:\Util\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-IE:\SourceTree\sloeber\MyESP8266/Release/core" -c -Wall -Wextra -g -ggdb -Og -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -DDEBUG -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DDEBUG_ESP_SSL -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_HTTP_SERVER -DDEBUG_ESP_UPDATER -DDEBUG_ESP_OTA -DDEBUG_TLS_MEM -DF_CPU=80000000L -DLWIP_OPEN_SRC  -DARDUINO=10802 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_ESP12\"" -DESP8266  -I"C:\Util\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266" -I"C:\Util\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\variants\nodemcu" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"
> In file included from C:\Util\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/Arduino.h:38:0,
>                  from ..\sloeber.ino.cpp:7:
> C:\Util\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/esp8266_peri.h:24:21: fatal error: c_types.h: No such file or directory
>  #include "c_types.h"
>                      ^
> compilation terminated.
> subdir.mk:31: recipe for target 'sloeber.ino.cpp.o' failed
> make: *** [sloeber.ino.cpp.o] Error 1
> 

According to the building message, it seems that somewhere hard corded the sdk path as 

> {sloeber}/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include
> {sloeber}/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include

as I have installed 2.4.1, so the sdk files cannot be found.

I have try to search the preferences settings, but I cannot found any entries which has marked 2.3.0.
The Include Folders in Settings of  C/C++ Build are correctly point to 2.4.1 folders, not 2.3.0.  But the used for building the project is 2.3.0.

In fact, I have only installed 2.4.1 so far, I don't why it has generated the include folder to 2.3.0.

May I know how I can change the sdk folder from 2.3.0 to 2.4.1?
Or is Jantje 1.0.0 only work for 2.3.0?  So that I must install ESP8266 2.3.0 in order to use Jantje?

jantje, 2018-07-31:
Hi
This is not a sloeber issue but a https://github.com/jantje/hardware issue.
As you can see in the platform.txt version 2.3.0 is hardcoded. https://github.com/jantje/hardware/blob/master/jantje/ESP8266/platform.txt#L12
You can install 2.3.0 and 2.4.1 next to each other in Sloeber or you can modify the platform.txt.
I remember 2.3.0 was needed but then we found a simple to fix bug that allowed for 2.4.x but I can't remember much of it any more. Some googling may help
Jantje

Super169, 2018-07-31:
Thanks for the information.
For safety, I will try to install 2.3.0 for it.

closed 2018-07-31 by Super169