Sloeber/arduino-eclipse-plugin #1004: Installation failing - incorrect path to avr tools

sk8nfool, 2018-08-02:
I am getting 4 errors when I start sloeber for the first time. The errors are all similar to this

Program "/home/braun/bin/sloeber/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" not found in PATH

For some reason sloeber has decided that "/home/braun/sloeber" is the installation directory for sloeber. It is not - "/home/braun/sloeber" is a sym-link to the executable "sloeber-ide" left over from a previous attempt at installing sloeber that I accidentally destroyed.

jantje, 2018-08-02:
Is this the plugin or the product?
is there /home/braun/sloeber in your eclipse.ini?

sk8nfool, 2018-08-02:
product - new install no such file eclipse.ini

I have eclipse installed as a standalone(not part of a package) but it
doesn't work at the moment - some kind of java error that I haven't
researched.

I will try uninstalling eclipse and try again.

BTW - I removed the version I'd installed and removed sloeber related stuff
from .cache. I then expanded the tar file in /opt (creating /opt/sloeber)
and changed ~/bin/sloeber-ide to point at /opt/sloeber/sloeber-ide. Firing
this up got me to the welcome screen (after "completing the download").
This was an improvement. I'd changed the previous install folder from
~/.local/share/sloeber/sloeber to
~/local/share/slober/V4.2_linux64.2017-11-20_02-16-52. Perhaps the path was
too long?

jantje, 2018-08-02:
Path length issues are a windows thing.
I have no clue what went wrong because you started from the product.

sk8nfool, 2018-08-03:
That's my understanding too.
Where does slobber keep these paths?

jantje, 2018-08-03:
Sloeber asks the installation path to Eclipse. 
https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/common/ConfigurationPreferences.java#L117
I have seen "wrong" eclipse home locations due to information in the .ini file next to the eclipse.exe. But the product should never have this issue as .ini is in the tar and does not contain that info.

jantje, 2018-08-03:
unless ... oomph is at play

jantje, 2018-08-24:
Did you make any progress.

jantje, 2018-09-08:
No update fro 15 days so closing issue

closed 2018-09-08 by jantje