Sloeber/arduino-eclipse-plugin #1007: Interrupts causing PSRAM init failed

RA5040, 2018-08-05:
I've tried the GPIOInterrupt example and also this one: https://techtutorialsx.com/2017/10/07/esp32-arduino-timer-interrupts/

Both result in a "[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!" error.  This issue is reported here: https://www.esp32.com/viewtopic.php?t=2031, with a workaround, which is to use the 'xtensa-1.22.x-ext_ram_patch'.  I have no idea if this is relevant to the arduino-eclipse plugin or not (and if it is how to implement it).

I have run the gpio example in the esp idf and it runs fine, so my hardware would appear to be OK.

RA5040, 2018-08-05:
It could be that the reason is I don't have the esp boards installed. I have followed your video carefully (https://www.youtube.com/watch?v=szRrvJPvQz4&feature=youtu.be).  The ESP32 is still not showing.

How do I add the esp32 json?

jantje, 2018-08-05:
Installing esp32 this way will not make it visible in the boards manager as it is not managed by the boards manager :-)
esp32 recently released a json file. https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls

RA5040, 2018-08-05:
It's these "little things" that catch one out ... Ignorance is not always bliss!

Anyway, I went into "Arduino/Preferences/Third party index url's" and added this line from your link: https://dl.espressif.com/dl/package_esp32_index.json

Now the ESP32 does show in the Platforms and Boards, so the that's a good step in the right direction :)

Unfortunately I still have the same issue with the interrupt routine. To explain (to the extent that I understand it) the EPS32-WROOM uses an RAM external memory which is called the PSRAM. Some code, in particular interrupt service routines, must be run in this memory.  It appears that there is/was some problem with the implementation. The workaround appears to be to use the patch I mentioned above.

This link gives more information on the patch: https://www.esp32.com/viewtopic.php?t=2967, specifically:
sudo apt-get install git wget make libncurses-dev flex bison gperf python python-serial
sudo apt-get install gawk gperf grep gettext python python-dev automake bison flex texinfo help2man libtool libtool-bin g++ libncursesw5-dev
cd ~/esp
git clone -b xtensa-1.22.x-ext_ram_patch https://github.com/espressif/crosstool-NG.git
cd crosstool-NG
mkdir .build
cd .build
wget http://loboris.eu/ESP32/crosstool-NG_psram_.build_tarballs.tar.xz
tar -xf crosstool-NG_psram_.build_tarballs.tar.xz
rm crosstool-NG_psram_.build_tarballs.tar.xz
cd ..
./bootstrap && ./configure --enable-local && make install
./ct-ng xtensa-esp32-elf
./ct-ng build
chmod -R u+w builds/xtensa-esp32-elf

This may be the fix needed ... but I don't know enough about the Eclipse-Arduino environment to understand how to install the patch so that it will be implemented in the plugin.

There is also a flag for the gcc compiler which may be relevant:
http://esp-idf.readthedocs.io/en/latest/api-guides/external-ram.html

jantje, 2018-08-05:
I think the best way forward is asking the esp32 community if the json installed boards have this fix or not.
And if they don't what to do.

RA5040, 2018-08-05:
I've run the program using Arduino IDE and it works without the PSRAM init issue, so this has something to do with the Eclipse Arduino and not the board. Also, as I mentioned before, I can also run interrupt routines using the esp idf.

RA5040, 2018-08-05:
I've posted a question in the esp32 forum, but it has to be approved by a moderator and the moderators seem to be on holiday because I am waiting over a week for approval on my first post.  If you have access, could you post it please?  Here is the post:

"I am testing an ISR on an  ESP32-DevKitC equivalent (JZK ESP-32S ESP32) using Eclipse Arduino. I am running on Windows 10.

I've run the test programs successfully using the Arduino IDE (for example this one: https://techtutorialsx.com/2017/10/07/esp32-arduino-timer-interrupts/).  I have also run the gpio program successfully from the esp idf.

With the Eclipse Arduino plugin I get the "[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!" error at startup.

I've reported the issue to Jantje here: https://github.com/Sloeber/arduino-eclipse-plugin/issues/1007.  He suggested that I check with this forum to see if this is a known issue and if there is a solution.

I thought that the issue might have to do with the PSRAM problem reported here: https://www.esp32.com/viewtopic.php?t=2031 and here: https://www.esp32.com/viewtopic.php?t=2967.  However I have no idea if this is the problem and if it is, how should I go about fixing it on the Eclipse Arduino."

RA5040, 2018-08-05:
The program does in fact run. I get the initialization error and then it works. So at least I can get on with things while we sort out the issue. 

As the initialization error is only happening with the plugin it must have something to do with the code generation.  It would be useful if someone else could try this to rule out the fault being with my setup.

jantje, 2018-08-14:
Before you can say it is with Sloeber you have to be sure all parameters are the same.
That is platform and versions
Board compilation settings

jantje, 2018-08-14:
In project properties->arduino you should have the same settings as in arduino bords menu.
Also check the boards manager. using the same json file and the same version of the platform?

RA5040, 2018-08-14:
Thanks ... the Arduino Core level debug was set to None whereas it was set to Verbose on Sloeber. I am now getting the same PSRAM init failed! warning on the Arduino flash.

So unless someone here has anything to add, it seems that this is just a bit of sloppy reporting from the initialization code.

Thanks for your time!

jantje, 2018-08-14:
A good place to show your thanks is http://eclipse.baeyens.it/donate.shtml

closed 2018-08-14 by jantje