Sloeber/arduino-eclipse-plugin #1008: GLIB-Critical errors when sloeber starts

sk8nfool, 2018-08-06:
I have installed sloeber from V4.2_linux64.2017-11-20_02-16-52.tar.gz on Ubuntu 18.04. When I start it I get a bunch of errors "... g_base64_encode_step: assertion 'in != NULL' failed". I searched for this message and found that if I start sloeber with

SWT_GTK3=0 sloeber-ide

then these messages are suppressed. What I don't know is what the total impact is. The errors don't seem to be detrimental but ....

jantje, 2018-08-06:
From what I see it is a plugin start order issue.

sk8nfool, 2018-08-11:
how is that determined?

jantje, 2018-08-11:
>how is that determined?

It is complicated (read I don't really know myself). Basically each and every plugin?/product?/feature gives a loading sequence of the "used plugins" like here https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.product/arduino.product#L354
I've been told "If it works, keep your hands of it".

jantje, 2018-09-08:
Can this issue be closed?

closed 2018-09-28 by jantje