Sloeber/arduino-eclipse-plugin #1009: minor error in sloeber.ino.cpp

sk8nfool, 2018-08-06:
I have installed sloeber using V4.2_linux64.2017-11-20_02-16-52.tar.gz on Ubuntu 18. I added Adafruit's AVR boards and created a default project called test. I looked at the sloeber.ino.cpp file and noticed that the

#include "Arduino.h"

line was duplicated. This isn't a big problem but isn't technically correct.

How is this file generated? The comment says it's automatically generated. Is there a template?

jantje, 2018-08-06:
The double include arduino.h is kind of like "by design".
That is .. as .ino files do not need to include arduino.h; arduino.h is added by default.
If there is a arduino.h in the ino file arduino.h will be there 2 times
This is not a problem due to the include guards.

closed 2018-08-06 by jantje

sk8nfool, 2018-08-06:
Like I said - minor problem but now I understand why. Thanks.