Sloeber/arduino-eclipse-plugin #1010: Failed to download json

AlirezaSadri, 2018-08-09:
Hi,

I was trying to addSTM32duino in Sloeber and the adoption encountered the following error:

Failed to install STM32 Cores
Failed to download "https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/tools/CMSIS-5.3.0.tar.bz2".
Read timed out

Also it couldn't download the arm gcc and I solved it by simply downloading these tar.bz2 files from STM32duino repository (which took a while to initiate....hence the timeout) and putting them in the download folder of arduinoplugin. So the json reader didn't have to download them again. Maybe its nice to be able to set the timeout for downloading packages.

Best regards
Alireza

jantje, 2018-08-24:
>Maybe its nice to be able to set the timeout for downloading packages.

I guess this would be a java setting somewhere. But I assume there is somewhere a network issue on your system (did you allow java internet access?) or a very bad network connection.

Anyone any idea how the timeout can be set in java?

jantje, 2018-08-26:
>Do you think that can be it?

I would think not. give it a try.

klause, 2018-08-31:
I had the same issue.

Problem is read timeout is hardcoded in [PackageManager.java](https://github.com/Sloeber/arduino-eclipse-plugin/blob/801541d3f02a7016ec40607a47bef5718f03608a/io.sloeber.core/src/io/sloeber/core/api/PackageManager.java#L724):

`			conn.setReadTimeout(5000);`

klause, 2018-08-31:
Workaround is to manually download json files to arduinoPlugin folder and zip files to arduinoPlugin/downloads

jantje, 2018-08-31:
@klause 
Cool seems like you found where the timeout is set :-)
5 seconds is a lot and still not all that much. Any clue what would be a more acceptable value?

AlirezaSadri, 2018-08-31:
@jantje
Hi, Mine needed at least 15 seconds

jantje, 2018-09-01:
@AlirezaSadri 
Wow that is a lot. Sounds like it will be best to make it configurable.

klause, 2018-09-05:
@jantje 
I think the best strategy is to use a higher timeout like 30 seconds but first test to see if sun.net.client.defaultReadTimeout system property is set and if so use the value from this property. Also you have to give some attention to sun.net.client.defaultConnectTimeout

Not sure, but there is a way of downloading files from eclipse plugin code that is using ECF Filetransfer Provider and let eclipse manage timeout and other connection aspects.

jantje, 2018-09-19:
I'm going to increase the timeout to 30 seconds as I have seen some "failed to download jsons" as well.
If anyone has a better way to do this, please make a pull request.

klause, 2018-09-19:
> @klause
> It looks like you know much more about this than I do :-)
> Would you be willing to improve the timeout implementation?

I will try to help

jantje, 2018-09-28:
I'm running here with the 30 seconds timeout now. I still get the failed to download but...(maybe my brains trick me)... less

jantje, 2018-09-29:
Today I got a 206 error for a json file. This implies the files are downloaded in chuncks (which should be good for archives) but somewhere something is not right

jantje, 2018-11-23:
Getting the following error is pretty consistent now :-s

`Failed to download url http://www.leonardomiliani.com/repository/package_leonardomiliani.com_index.json error code is: 206`

rlogiacco, 2018-12-07:
I have been having that response code for the past say 4 months. The 206 is actually not °an error° as it is quite a common technique.
Can you pinpoint me to where this download is performed_ Iàll try to check if we are doing anything wrong...

jantje, 2020-04-21:
Does this issue still exists?

closed 2020-07-07 by jantje