Sloeber/arduino-eclipse-plugin #1013: Help! "#if__has_include("A.h")" can't be compiled

human890209, 2018-08-15:
I use a library which uses #if __has_include("A.h") before #include"A.h"
And this can't be compiled.

``
#if __has_include("FreeRTOSConfig.h")
#   include "FreeRTOSConfig.h"
#elif __has_include("configs/FreeRTOSConfig.h")  //
#   include "configs/FreeRTOSConfig.h"
#else /*FreeRTOS default define file*/
#   include "FreeRTOS/default/FreeRTOSConfig.h"
#endif
``

The compile shows:
> 
error: missing binary operator before token "("
error: missing binary operator before token "("
fatal error: FreeRTOS/default/FreeRTOSConfig.h: No such file or directory
> 

How to fix this?

liebman, 2018-08-15:
This is probably an issue with whatever library your using and not a Sloeber issue.  Its a C++17 extension and some compiler don't support that.

See http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2015/p0061r0.html

human890209, 2018-08-16:
Hi,
So it's not a standard function of C++17, so the compiler doesn't support that.
Is the solution just comment out them and #include "someLibrary.h" manually?

human890209, 2018-08-17:
Hi,
I tested the sketch with Arduino IDE, and it works.
So as Sloeber's target is to compile sketches as well as Arduino IDE.
Hope there will be a fix or setting instruction for this.

human890209, 2018-08-17:
And I found this Issue Report:
[https://github.com/Sloeber/arduino-eclipse-plugin/issues/988](url)

I think Sloeber doesn't get a fix for now... 
Hope there will be some good news soon.
I will close this issue for now.

closed 2018-08-17 by human890209

sk8nfool, 2018-08-17:
I'm using Sloeber V4.2_linux64.2017-11-20_02-16-52 and __has_include(...) works just fine. I installed the sloeber package (not the plug-in) if that makes any difference.

`
\#if __has_include ("Arduino.h")
 \#	include "Arduino.h"
 \#else
 \#	error Has include failed
 \#endif

 //The setup function is called once at startup of the sketch
 void setup()
 {
 // Add your initialization code here
 }

 // The loop function is called in an endless loop
 void loop()
 {
 //Add your repeated code here
 }
`

liebman, 2018-08-17:
What core are you using?  In most cases the core selection designates what compiler version is used.

sk8nfool, 2018-08-18:
good question. don't know how to find out. Not sure what "core" means: core
of eclipse, the plugin or what. The avr-g++ executable claims version 4.9.2