Sloeber/arduino-eclipse-plugin #1015: Remove "path too long" error at eclipse start (as an option)

Caceresenzo, 2018-08-24:
Hi,

Having this error everytimes i open eclipse is pretty anoying, are you considering make an option to "hide" it ? 

- Thanks

jantje, 2018-08-24:
move sloeber closer to the root of your hard drive?

Caceresenzo, 2018-08-24:
rude.

I asked that question because I can't move my Eclipse folder closer of the root....

Are you planning anything ?

jantje, 2018-08-24:
>rude.

So is:
1 not doing some efforts to investigate the issue. 
2 Not supporting the person spending lots of time to develop
3 maintain 
4 support the tool you are using for free
5 feeling the need to answer rude when someone answers your question for free (again)
In other words.. if you had bothered to support the project or even bothered to tell you can't move closer to the root you may have received a nicer answer.
If you don't care about all of that and still want a nicer answer ask the question on patreon. The more support, the nicer I become.

closed 2018-08-24 by jantje

Caceresenzo, 2018-08-24:
wtf

wasn't your previous anwser sarcasm ?

I personnaly use Microsoft OneDrive to sync all my data, so no i can't put my eclipse folder closer to my hard drive;
2nd: I am french, and i was thinking "rude" was meaning something like a french expression "heavy, weight" (meaning that the anwser was not what i anwser)
3rd: I am a developper myself and i personnaly love helping other;
4th: Your plugin ask the user every 10/20 lunch to help a patreon.... So you must be a least dispose to help other with your unwanted ads

Caceresenzo, 2018-08-24:
Support suck here

I never insult people on the internet but hey, there a start for everything!

I re-compiled your code with the error code removed, and here it is:
[http://www.mediafire.com/file/zt8fcv088n82rsm/features.zip/file](http://www.mediafire.com/file/zt8fcv088n82rsm/features.zip/file)

To install it, go to Help > Install Software
- Add
- Click Archive and select the downloaded archive
- Click ADD
- Uncheck everything "Show only the lastest versions of available software", "Group items by category" and "Hide item already installed" and it should appear
- Install, wait, and your done with this message