Sloeber/arduino-eclipse-plugin #1017: Platform manager can't download SAM M3

p-dekker, 2018-09-07:
When I use the platform manager to download the SAM m3 platform. It always fails for 1.6.11 and 1.6.10. (1.6.9 does work and 1.6.11 and 1.6.10 do work in Arduino IDE)

It fails on downloading http://downloads.arduino.cc/cores/sam-1.6.10.tar.bz2 or http://downloads.arduino.cc/cores/sam-1.6.11.tar.bz2

I can manually download both files. So it's not network issues. I think the length and/or SHA digest in the .json file is incorrect and Sloeber is aborting on installing it. 

best
Patrick

jantje, 2018-09-07:
may be related to #1010 
But as you do not provide the error it is hard to tell.
As a workaround drop the downloaded files in [sloeber]/arduinoplugin/downloads
This way Sloeber will not try to download the files but start from the files you provided.

p-dekker, 2018-09-10:
Both give this error. 

Failed to install Arduino SAM Boards (32-bits ARM Cortex-M3)
Failed to extract tar.bz2. "/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/downloads/sam-1.6.11.tar.bz2".
unexpected end of stream

I will try to have a look at the code and see what's wrong.

jantje, 2018-09-10:
I guess the download of /sam-1.6.11.tar.bz2 failed (but is partially available) and the only way out of this situation is to delete the file in the downloads folder

jantje, 2018-09-19:
Any update on this?

closed 2018-09-28 by jantje