Sloeber/arduino-eclipse-plugin #1018: Installation on macOS not possible

Betacarotin, 2018-09-08:
Hi! My installation stops already after 1 second with this here: 

Failed to find a Main Class in "/private/var/folders/fl/zzmgphz550j6q6mxz1xb0tkw0000gn/T/AppTranslocation/F94DBF8A-53DE-4BA2-9BA8-DDE26CE4751B/d/sloeber.app/Contents/MacOS//../Eclipse/plugins/org.eclipse.equinox.launcher_1.5.0.v20180512-1130.jar".

jantje, 2018-09-08:
What java do you run?

Betacarotin, 2018-09-08:
I have absolute no clue.. how do I know? Its kind of ”out of the box macOS” with standard Arduino IDE, xCode and some other stuff on. Had never anythong like Eclipse running on it.

jantje, 2018-09-08:
java -showversion tells you the version.
You need at least version 8 (1.8 says showversion)
You can copy the java folder from arduino IDE under Sloeber and restart. That "should work"

closed 2018-09-28 by jantje