Sloeber/arduino-eclipse-plugin #1020: Parameters of espota not correct when not in main cpp file

Bascy, 2018-09-15:
I'm experimenting with OTA for our ESP32. I know the ESP32 is not supported but this issue is not specifically for the ESP32.

If I use a test script with every configuration code of the ArduinoOTA object then the call of espota is corerct, including the correct portnr and password.

If I move the code that set these configuration settings to a different source file, i.e. one containing a container object for the ArduinoOTA, then espota is called with the wrong (default) parameters.

Is there a way to specify the parameters of espota other then setting the ArduinoOTA configurations in the main source file?

jantje, 2018-09-15:
I do not understand what you are talking about.
First of there is a esp32 json file you can use. https://dl.espressif.com/dl/package_esp32_index.json

> If I move the code that set these configuration settings to a different source file, i.e. one containing a container object for the ArduinoOTA, then espota is called with the wrong (default) parameters.

I have no clue what you are talking about. Can you provide an example of where the wrong parameters are used and tell me what you think are the correct parameters?

Bascy, 2018-09-15:
I added example code to the orignal post

closed 2018-09-15 by jantje

Bascy, 2018-09-15:
That's what I afraid of ... Its not exactly clear to me why you should read those settings from the setup code in the main sourcefile, as that severly limits development possibilities. It will also not work if you change the portnr in the code, as the operational script still has the old portnr ...

Can I override the settings used, or use other settings (like -P) by defining them somewhere else? Like boards.txt or an other settings file?

jantje, 2018-09-15:
Just make a standard eclipse external launch configuration with the port an password you want
Don't forget to also configure the build tab.
[image]
[image]

Note that I have been where you are going and in the end I simply put the commands back in setup. I found that the disadvantages to moving this code in a class constructor or initializer did not outweigh the disadvantages.