Sloeber/arduino-eclipse-plugin #1028: AWS IoT ESP32: error MBEDTLS_CONFIG_FILE

OllisGit, 2018-10-04:
Hi,
I used this tutoriual: https://www.instructables.com/id/Secure-IOT-With-AWS-and-Hornbill-ESP32-Using-Ardui/  to connect my esp32 with the aws cloud.
Everthings works fine, if I use the Arduino IDE 1.8.7, but if I switch wo sloeber the programm doesn't work.
OS: Windows 7

I received the following error:
- several *.h files located under /Expressif-ArduinoESP32/tools/sdk/include/mbedtls/mbedtls were effected
- error-message: `#include expects "FILENAME" or <FILENAME>`
- e.g. platform.h, debug.h, pk.h, ssl.h, ....
```
#if !defined(MBEDTLS_CONFIG_FILE)
#include "config.h"
#else
RED-MARKED: #include MBEDTLS_CONFIG_FILE RED-MARKED
#endif
```
Any idea how to solve this or how to analyse the bug deeper?

BR
Olli

jantje, 2018-10-04:
This is probably related to command line high tech (which only works in windows in some linux simulators and Sloeber works natively) and not using 
```

#define VAL(str) #str
#define TOSTRING(str) VAL(str)
```

Which should be used to convert defines into strings in C/C++

OllisGit, 2018-10-04:
first: wow, thanks for the fast response!

Unfortunately, I didn't understand your answer.
- Should I run "sloeber" on a linux-system, could that be the solution?
- Or I do the pre-processing by hand: patch every include: replace MBEDTLS_CONFIG_FILE with mbedtls/esp_config.h
see https://github.com/espressif/arduino-esp32/issues/1327

jantje, 2018-10-04:
I don't have time to try this out but easiest way out is something like this

```
#if !defined(MBEDTLS_CONFIG_FILE)
#include "config.h"
#else
#define VAL(str) #str
#define TOSTRING(str) VAL(str)
#include TOSTRING(MBEDTLS_CONFIG_FILE) 
#endif
```

OllisGit, 2018-10-04:
That solves the problem!!!!
Thanks a lot, now I can work with your GREAT IDE again!!!!

closed 2018-10-04 by OllisGit

jantje, 2018-10-04:
Think about this next time you see http://eclipse.baeyens.it/remind.php

Bascy, 2018-12-16:
I made a pull-request with a fix for this for the usage of ARDUINO_VARIANT in the ESP32mDNS library in arduino_esp32 project and it was accepted

jantje, 2018-12-16:
@Bascy cool.
do you have a link?

Bascy, 2018-12-16:
https://github.com/espressif/arduino-esp32/commit/4d3f6caa0d23b741f668f3a72c18a93227523bd8

mikemoretti3, 2019-05-26:
I'm still seeing this issue.  I had to edit the Environment and change the A.COMPILER.CPREPROCESSOR.FLAGS where it says -DMBEDTLS_CONFIG_FILE\="mbedtls/esp_config.h" to -DMBEDTLS_CONFIG_FILE\="""mbedtls/esp_config.h"""   note the triple quotes.

executer-uno, 2019-12-19:
Me too. Clean fresh Install. Fresh Ububntu system.
I was managed to make it work with change DMBEDTLS_CONFIG_FILE to:
-DMBEDTLS_CONFIG_FILE="<mbedtls/esp_config.h>"

PS: for me as newbe it was not obvious. You can find Enviroment variables in Project/Properties/C/C++_Build/Enviroment
There is a big table where you can find A.COMPILER.CPREPROCESSOR.FLAGS variable. It contains a lot of definitions, be carefull to edit DMBEDTLS_CONFIG_FILE only.

jantje, 2020-05-21:
I think this is fixed. If not please provide the json and board settings that fail.

punj, 2021-04-24:
This fixed the issue in Eclipse. However I am not sure if Arduino IDE can still compile?

jantje, 2021-04-24:
@punj 
You should not share arduino IDE and Sloeber packages. So sloeber workarounds/fixes at the level of platforms should never hit Arduino IDE