Sloeber/arduino-eclipse-plugin #1029: Building the product sdk does not work with latest

jantje, 2018-10-10:
When building the product with the command line as specified in the read.me
mvn clean verify -PSDK.latest -DskipTests=true
I get the following output

```
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO]
[INFO] io.sloeber.parent 4.3.1-SNAPSHOT ................... SUCCESS [  0.469 s]
[INFO] io.sloeber.application 4.3.1-SNAPSHOT .............. SUCCESS [  3.368 s]
[INFO] io.sloeber.core 4.3.1-SNAPSHOT ..................... SUCCESS [  3.188 s]
[INFO] io.sloeber.core.nl1 4.3.1-SNAPSHOT ................. SUCCESS [  0.328 s]
[INFO] io.sloeber.ui 4.3.1-SNAPSHOT ....................... SUCCESS [  1.937 s]
[INFO] io.sloeber.ui.nl1 4.3.1-SNAPSHOT ................... SUCCESS [  0.250 s]
[INFO] io.sloeber.feature 4.3.1-SNAPSHOT .................. SUCCESS [  0.516 s]
[INFO] io.sloeber.product 4.3.1-SNAPSHOT .................. SUCCESS [01:24 min]
[INFO] io.sloeber.updatesite 4.3.1-SNAPSHOT ............... SUCCESS [  5.563 s]
[INFO] io.sloeber.tests 4.3.1-SNAPSHOT .................... SUCCESS [  3.047 s]
[INFO] root 2.2.0 ......................................... SUCCESS [  1.343 s]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 02:25 min
[INFO] Finished at: 2018-10-10T18:21:42+02:00
[INFO] ------------------------------------------------------------------------
[WARNING] The requested profile "SDK.latest" could not be activated because it does not exist.

C:\Users\HP\Documents\GitHub\arduino-eclipse-plugin>mvn clean verify -PSDK.latest -DskipTests=true
```
Note the warning and the fact product.sdk is not in the list.
There is also no target folder created in the target.sdk folder.
This is when using a new machine (installed github cient and maven and pulled the sloeber code today to take a machine to Rome)

wimjongman, 2018-10-12:
mvn clean verify -PSDK.latest -DskipTests=true

should be SDK,latest (comma)

jantje, 2018-10-15:
Thanks wim
Fixed the documentation

closed 2018-10-15 by jantje