Sloeber/arduino-eclipse-plugin #1030: OSX App icon is missing

mlasevich, 2018-10-15:
This is minor, all things considered,  but its been bugging me for a while now. For whatever reason, when you add the all-in-one Sloeber app to the Applications in OSX, it uses the default Apple "pencil/brush/ruler" icon instead of Sloeber's own icon (same in the dock). 

[image]

Once the app is fully running, icon in the dock is fine, but as soon as it shuts down it goes back to the default icon.  I thought it was just my machine acting weird but I just reloaded a fresh OS (10.14) and a fresh copy of Sloeber (4.3.1) and it is still doing the same thing.

I suspect it has something to do with sizes of icons, as when I look at it with Finder in list view (small icons) it shows proper Sloeber icon, If i change to Icon view (larger icons) it is the default icon.

jantje, 2018-10-15:
Thanks for letting us know.
As I don't own a mac.......would you mind helping to debug?

mlasevich, 2018-10-15:
I am happy to do whatever I can, but not being a mac developer, not exactly sure what I am looking for...

mlasevich, 2018-10-15:
Actually I think I have a clue and if I am right, it should be a simple fix. Because newer macs support higher resolution (Retina display and what not) you are supposed to include high resolution icons as well (see here: https://developer.apple.com/library/archive/documentation/GraphicsAnimation/Conceptual/HighResolutionOSX/Optimizing/Optimizing.html) but the icons include only have up to 128x128

jantje, 2018-10-15:
Thanks for the investigation. Seems like we have a missing icon format issue here.
But we also have a os comminication issue :-).
I can open the zip but my (windows) system doesn't know what to do with icns files.
Anyway, below are the formats currently supported by eclipse and provided by Sloeber.
[image]
I can get the 64x64 and 128x128. would that do the trick?

mlasevich, 2018-10-15:
Not sure, the doc I found claim you need to have up to 512, and for each  resolution they wanted an `@x2` version (which appears to be a double resolution image, so 16x16@x2 is really 32x32 - not sure why they want that, but still)

As far as icns file - it seems to be just a collection of icons in a single file - mac has a command "iconutil" which allows you to extract/recompress them, but inside they are just png's with specific names (like `icon_128x128.png` and `icon_128x128@2x.png`) so I am guessing the tool above generates that. That said, I am not sure how much value is the file I uploaded is, as all I did was add the missing pngs by scaling up the existing ones i.e. the quality is probably less than stellar.

The icon.icns file I had 16, 32, 48 and 128 pixel versions  and  no `@x2`'s and seems to match the file found here: https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.product.sdk/launch/icon.icns

Really wish there was a "cloud" rent-a-mac service where you can test things on a mac when you do not have one...

mlasevich, 2018-10-15:
I am no artist myself, but I think SVG should produce scalable images - I will take a stab at it tonight.

mlasevich, 2018-10-16:
Ok, not being an artist but coder, threw together this using imagemagik/rsvg:

```
#!/bin/bash

src="logo_256.svg"
dir="icon.iconset"

rm -rf "${dir}"
mkdir -p "${dir}"

function makePNG(){
  local -i px=${1}
  rsvg-convert -h ${px} -w ${px} ${src} > "${dir}/icon_${px}x${px}.png"
  rsvg-convert -h $(( 2 * px ))  -w $(( 2 * px)) ${src} > "${dir}/icon_${px}x${px}@2x.png"
}

makePNG 16
makePNG 32
makePNG 128
makePNG 256
makePNG 512

iconutil -c icns icon.iconset
```

result: 
[icon.icns.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/2481349/icon.icns.zip)

Seems ok to me and much smaller than my first attempt...

jantje, 2018-10-16:
I have been trying to get the files 64X64 and 128X128 from the icons.icns file but failed.
Can you add these here?
That way I can add the to the repository and the dialog provided above.

jantje, 2018-10-16:
@wimjongman what do you think about the icons in the SDK product.
I would assume it needs other icons. On the other hand...is it worth the investment?

mlasevich, 2018-10-16:
64 pixel is not required, AFAIK, but I will create it too.

Attached is a zip of both exploded and compressed icon set INCLUDING 64px and 48px (which are not required according to apple docs above)

[iconset.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/2484172/iconset.zip)

jantje, 2018-10-16:
Thank you very mutch.
I didn't ask the png files for apple OS but for the eclipse product.
Not sure now whether to use the full new -not al needed- icon.icns or stay with the one you provided before.

@mlasevich For your info the product.sdk is the java sdk to make the sloeber product. As to me the product.SDK should have different icons because it is not Sloeber. As wim is the most eclipse knowledgable person I know and wim made the product.SDK I feel it is up to him to say what to do.
The commit you see above replaced the icon.icns for sloeber and should be available on the nightly update site from tomorrow onwards.
Now I will make a commit to add the png files for eclipse.

mlasevich, 2018-10-16:
FYI - `rsvg-convert` (should be available on all major OS's)  will convert, on command line,  the SVG to PNG for you in any way you need , and since SVG's are vector, you should be able to scale them to any size without quality loss. Usage is `rsvg-convert -h XX -w XX file.svg > file.png` where XX is the width/height of the output image' . Given that you have the SVG version of the logo, should be easy enough to generate anything you need (if I did not include it)

The file I posted earlier, had the 10 png's Apple asked for  (16, 32, 128, 256, and 512 in both standard and "2x hi-res" formats) so it should be enough. The last one I added into the zip for sake of completeness, not being sure why you asked for the PNGs. I do not think you need it.

Also, I am not too familiar with how eclipse based products are generated, but I would assume that the icon pack in sdk is there to be included in the output of SDK build (i.e. product itself) - rather than for the SDK itself - no?

jantje, 2018-10-18:
>Also, I am not too familiar with how eclipse based products are generated, but I would assume that the icon pack in sdk is there to be included in the output of SDK build (i.e. product itself) - rather than for the SDK itself - no?

Nit sure what you mean. As far as I know;  The product.SDK is a copy of the product and then modified to contain all the development packages.
This is why it contains the same icons. Wim made the product.SDK to make it easier for people to contribute. I'm using the product because I want to use the same thing as the end user.

Have you been able to upgrade to the nightly to see if it is fixed now?

mlasevich, 2018-10-18:
I  misunderstood. I thought you used SDK to generate the distributable "binaries". I have not tried the nightlies yet as I just dropped the iconset into my existing install, but I will try it now

closed 2018-10-18 by jantje

wimjongman, 2018-11-09:
The Icons look very nice. Please also add them to the SDK. Please note that the icons you specify there are only the launcher icons. NOT the product icons. I'm not sure how Eclipse handles the @2x icons for hdpi displays.