Sloeber/arduino-eclipse-plugin #1031: Sloeber use of webkitnetworkprocess

sk8nfool, 2018-10-15:
I keep running into a situation where I can't open the serial port because its busy. Looking for the device using "lsof | grep ttyA" shows that WebKitNetworkProcess has /dev/ttyAxxxx open. It (WebKit...) also has a bunch of files in my workspace open so I'm assuming either the sloeber plugin or eclipse (I'm using your "product" version 4.3.1) is starting WebKitNetworkProcess.

This doesn't happen right away. I haven't been able to identify when or what triggers this. The parent process is "1" which typically means the parent has died sometime after starting and abandoning the child.

WebKit... doesn't terminate if I close the Teensy loader but it does terminate when I leave sloeber. Nor does WebKit... start when I start Sloeber or compile my sketch or upload it to my teensy. In fact I can do many compile/run cycles before WebKit... gets in the way.

jantje, 2018-10-15:
I have no idea where this WebKitNetworkProcess comes from.
But given your last paragraph; It may be related to the changes Teensy Paul did in regards to upload.
Do you use Teensyduino 1.42 or later?
https://www.pjrc.com/blog/

sk8nfool, 2018-10-16:
I installed Teensyduino 1.44 onto Arduino 1.8.7
I searched for "WebKit" in the Teensy loader executable and found nothing. I'll start an strace looking for process control calls and see if it's guilty.

sk8nfool, 2018-10-17:
BTW - WebKitNetworkProcess is part of libwebkit2gtk-4.0-37 - still researching why this is installed.

sk8nfool, 2018-10-29:
I give up - hasn't happened again when I have an strace running.

closed 2018-10-29 by sk8nfool