Sloeber/arduino-eclipse-plugin #1034: Can't enter path for serial port in serial monitor

sk8nfool, 2018-10-29:
I have several devices that I use with Sloeber. Occasionally I have several connected to my system. Because by default Linux dynamically assigns the serial port names I found it difficult to determine which of the /dev/ttyACMx device paths belong to which device. So I created udev entries that create symbolic links to the ttyACMx device names that are unique according to the device's serial number. This works quite well. I can type in the device path (e.g. /dev/ttyARD0) when configuring the board in the project "properties/arduino" page but not when selecting the port in the serial monitor. Both need to let me enter arbitrary port paths.

jantje, 2018-10-29:
If you have your project selected and click the serial monitor in the toolbar the serial monitor will open using the baud rate used in the project in Serial.Begin and the com port of the upload.
Lots easier than manually selecting it.

sk8nfool, 2018-10-30:
That's cool - thanks - but the path to the device in the monitor is fixed.
It won't let me type in an arbitrary path where the Properties-Arduino does.

jantje, 2018-10-30:
>but the path to the device in the monitor is fixed.

yes it is. Feels like we have gone through this before #929

jantje, 2018-11-23:
tested works now

closed 2018-11-23 by jantje