Sloeber/arduino-eclipse-plugin #1035: When installing new platforms all json files need to be uptodate

jantje, 2018-11-06:
Reported by Patron
When installing a new platform it may be that the json file is referencing a tool in another json file which may not be referenced in the file on your system because it is "not the latest version"

Workaround:
Make sure you have the latests versions on your system by activating arduino->preferences->arduino->Third party index url's->update local json files(including libraries) and restarting Sloeber.

Solving the issue when it happened.
Apply workaround . remove the platform from your system by unselecting the version from Arduino->preferences->platform and boards select apply and close.
Reinstall the platform by selecting it again.

More details on when this happened and what is going on
This happened when installing "Adafruit SAMD Boards"  version 1.2.6 which uses tool 
```
            {
              "packager": "arduino",
              "name": "bossac",
              "version": "1.8.0-48-gb176eee"
            },
```
This tool is defined in package_index.json but is not used by any of the platforms in the package_index.json. 

```
        {
          "name": "bossac",
          "version": "1.8.0-48-gb176eee",
          "systems": [
           .....
```
            
This way the "default workaround" "install the platform using that tool" will not work.

jantje, 2018-11-07:
Though the issue above is a problem in rare cases it is probably not what is gong on at the patron's site.
This because on my system (installed with all latest jsons) Sloeber uses bossac v1.7.0
I tried with the arduino IDE but as I do not have the board Arduino IDE stops after a failed reboot and does not tell me which version it planned to use.
I noticed this in the platform.txt.
```
tools.bossac.path={runtime.tools.bossac-1.7.0.path}
tools.bossac.cmd=bossac

tools.bossac.upload.params.verbose=-i -d
tools.bossac.upload.params.quiet=
tools.bossac.upload.pattern="{path}/{cmd}" {upload.verbose} --port={serial.port.file} -U {upload.native_usb} -i -e -w -v "{build.path}/{build.project_name}.bin" -R
...

# v1.8.0

tools.bossac18.path={runtime.tools.bossac-1.8.0-48-gb176eee.path}
tools.bossac18.cmd=bossac

tools.bossac18.upload.params.verbose=-i -d
tools.bossac18.upload.params.quiet=
tools.bossac18.upload.pattern="{path}/{cmd}" {upload.verbose} --port={serial.port.file} -U -i --offset={upload.offset} -e -w -v "{build.path}/{build.project_name}.bin" -R
```

So both bossac version 1.7 and 1.8 have a upload description.
The boards.txt states for feather MO 
`adafruit_feather_m0.upload.tool=bossac`
So bossac should be used -with feather mo- which translates to version 1.7 (like sloeber does)
The boards.txt also contains
`adafruit_trellis_m4.upload.tool=bossac18`
So when using the "Adafruit Trellis M4 (SAMD51)' bossac version 1.8 should be used.

I think the problem is that this platform needs 2 different versions of the same tool and can only specify 1 in the json file.
They opted to install only 1.8 and assume 1.7 is installed.
The patron installed Sloeber (which means the latest Arduino AVR at the time) and added adafruit SAMD.
At that point in time bossac V1.7 is not available to Sloeber at which point the upload fails because the full path is unknown resulting in {runtime.tools.bossac-1.7.0.path} being "" and as such sloeber uses  "/bossac" which is unknown on the system. 
I haven't tested this but I assume the same issue exists in the Arduino IDE (but you would need a "clean" system to test) 

Workaroud:
Install "Arduino SAMD Boards (32-bits ARM Cortex-M0+)" version "1.6.7". (which spoecifies bossac 1.7 as tool)
For all affected existing projects open project properties-Arduino->apply and ok
You can safely deinstall "Arduino SAMD Boards (32-bits ARM Cortex-M0+)" as this will not remove the installed tools.

meschli, 2018-11-07:
As an FYI, this was my issue and jantje nailed it.
I installed "Arduino SAMD Boards (32-bits ARM Cortex-M0+)" version "1.6.7" and subsequently my upload issues evaporated. ... I'm impressed.

closed 2018-12-03 by jantje