Sloeber/arduino-eclipse-plugin #1036: Updating Sloeber to 4.3.1.201810101819 fails because of Nebula version

Bascy, 2018-11-10:
When updating sloeber through "Check updates" in Eclipse the following error occurs, stating that my installed version of Nebula is 1.4.0.2018061814 and Sloeber want 1.4.0.201804281520

> Cannot complete the install because of a conflicting dependency.
  Software being installed: Sloeber, the Eclipse Arduino IDE (plugin version) 4.3.1.201810101819 (io.sloeber.feature.feature.group 4.3.1.201810101819)
  Software currently installed: Nebula Oscilloscope Widget 1.4.0.201808061814 (org.eclipse.nebula.widgets.oscilloscope.feature.feature.group 1.4.0.201808061814)
  Only one of the following can be installed at once: 
    Nebula Oscilloscope Widget 1.4.0.201808061814 (org.eclipse.nebula.widgets.oscilloscope.feature.feature.jar 1.4.0.201808061814)
    Nebula Oscilloscope Widget 1.4.0.201804281520 (org.eclipse.nebula.widgets.oscilloscope.feature.feature.jar 1.4.0.201804281520)

jantje, 2018-11-10:
If I remember correctly deinstalling nebula before upgrading worked fine.

Bascy, 2018-11-10:
That did the trick

wimjongman, 2018-11-14:
This is caused by the fact that the Nebula plugin is included and not referenced. I will take a look.

wimjongman, 2018-11-14:
Fixed with commit 090072ce82e7a9cafc126dfde6a037e91ee1ceab

closed 2018-11-14 by wimjongman