Sloeber/arduino-eclipse-plugin #1037: Third party index url's - JSON name package.json not allowed

lecommander, 2018-11-11:
I just tried to add this json file:
https://github.com/esp8266/Arduino/blob/master/package.json

I'm getting the error message: 
json files should start with "package_" or "library_" https://github.com/esp8266/Arduino/blob/master/package.json is ignored

How can I get that file included or would it be possible to to allow also "package.json" and "library.json"

jantje, 2018-11-11:
Your url is wrong. it should be
https://raw.githubusercontent.com/esp8266/Arduino/master/package.json

closed 2018-11-11 by jantje

lecommander, 2018-11-11:
Thank you for the correction of the url.
Unfortunetely the issue remains - error log:

> json files should start with "package_" or "library_" https://raw.githubusercontent.com/esp8266/Arduino/master/package.json is ignored

I think this is caused because file name is not **package_**xyz.json - as the error log indicates.