Sloeber/arduino-eclipse-plugin #1043: Debugger times out when launched from within sloeber, yet command line gdb is fine

Crazor, 2018-11-22:
I'm trying to debug ESP8266 code with sloeber 4.3 on macOS Mojave.

Right now, I run some minimal code on the device, using the gdbstub from the official esp8266 package (not with [jantje/hardware](https://github.com/jantje/hardware/)), just to remove as much uncertainty as possible. The code:
```
#include "Test.h"
#include <GDBStub.h>

void setup()
{
	Serial.begin(115200);
	*((int *)0) = 0; // crash
}

void loop()
{
}
```

When I run xtensa-lx106-elf-gdb from the command line, it connects successfully and displays the line the processor crashed on. When I try to debug from within sloeber, the debugger never launches, but instead hangs at "96% Configuring gdb".

I enabled the gdb trace console in sloeber's settings, and I see the following behaviour:
```
257,226 2-gdb-version
257,228 ~"GNU gdb (crosstool-NG 1.20.0) 7.5.1\n"
257,231 ~"Copyright (C) 2012 Free Software Foundation, Inc.\n"
257,231 ~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is fre\
e software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitt\
ed by law.  Type \"show copying\"\nand \"show warranty\" for details.\n"
257,231 ~"This GDB was configured as \"--host=x86_64-build_apple-darwin14.1.0 --target=xtensa-lx106-\
elf\".\nFor bug reporting instructions, please see:\n"
257,231 ~"<http://www.gnu.org/software/gdb/bugs/>.\n"
257,231 2^done
257,231 (gdb) 
257,237 3-environment-cd /Users/crazor/Documents/sloeber-workspace/Test
257,239 3^done
257,239 (gdb) 
257,240 4-gdb-set breakpoint pending on
257,241 4^done
257,241 (gdb) 
257,242 5-gdb-set detach-on-fork on
257,243 5^done
257,243 (gdb) 
257,245 6-enable-pretty-printing
257,248 6^done
257,248 (gdb) 
257,248 7-gdb-set python print-stack none
257,250 7^done
257,251 (gdb) 
257,251 8-gdb-set print object on
257,253 8^done
257,253 (gdb) 
257,253 9-gdb-set print sevenbit-strings on
257,255 9^done
257,255 (gdb) 
257,255 10-gdb-set host-charset UTF-8
257,256 10^error,msg="Undefined item: \"UTF-8\"."
257,257 (gdb) 
257,257 11-gdb-set target-charset UTF-8
257,258 11^error,msg="Undefined item: \"UTF-8\"."
257,258 (gdb) 
257,259 12-gdb-set target-wide-charset UTF-32
257,260 12^error,msg="Undefined item: \"UTF-32\"."
257,260 (gdb) 
257,260 13source gdbinit
257,261 &"source gdbinit\n"
257,262 13^done
257,262 (gdb) 
257,263 14-gdb-set target-async off
257,264 14^done
257,264 (gdb) 
257,264 15-gdb-set auto-solib-add on
257,265 15^done
257,265 (gdb) 
257,268 16-file-exec-and-symbols --thread-group i1 /Users/crazor/Documents/sloeber-workspace/Test/Re\
lease/Test.elf
257,280 16^done
257,281 (gdb) 
257,281 17-target-select remote /dev/cu.SLAB_USBtoUART
278,379 ~"Ignoring packet error, continuing...\n"
278,381 &"warning: unrecognized item \"timeout\" in \"qSupported\" response\n"
```

I added `set remotelogfile log.txt` to the gdbinit, and when ran from within sloeber, the logfile generated is:
```
w +$qSupported:multiprocess+;qRelocInsn+#2a
r <Timeout: 2 seconds>
w $qSupported:multiprocess+;qRelocInsn+#2a
r <Timeout: 2 seconds>
w $qSupported:multiprocess+;qRelocInsn+#2a
r <Timeout: 2 seconds>
w $qSupported:multiprocess+;qRelocInsn+#2a
r <Timeout: 2 seconds><Timeout: 2 seconds>
w -
r <Timeout: 2 seconds>
w -
r <Timeout: 2 seconds>
w -+$Hg0#df
r <Timeout: 2 seconds>
w $Hg0#df
r <Timeout: 2 seconds>
w $Hg0#df
r <Timeout: 2 seconds>
w $Hg0#df
```

When running gdb from commandline, the log of a successful connection contains:
```
w +$qSupported:multiprocess+;qRelocInsn+#2a
r +$swbreak+;hwbreak+;PacketSize=255#6b
w +$Hg0#df
r +$#00
w +$?#3f
r +$T0b#e6
w +$Hc-1#09
r +$#00
w +$qC#b4
r +$#00
w +$qAttached#8f
r +$#00
w +$qOffsets#4b
r +$#00
w +$g#67
r +$2c282040b0ffff3f000000002105000000c2ff3f21050000ffffffff200000003cdc0040000000000000000000000000b4e9fe3f00000000d0daff3fbce9fe3fbc2420401600000000000000fada8cc20000000030000000#19
w +$qSymbol::#5b
r +$#00
w +$qTStatus#49
r +$#00
w +
c quit
w $qTStatus#49
r +$#00
w +$k#6b
r +
End of log
```

Right now I'm pretty much stuck. Any help would be appreciated!

Crazor, 2018-11-22:
Oh boy. I'm always amazed how much rubber-ducking helps to clear the mind. I've been fiddling with this all day and the better part of yesterday. Right after posting my findings, it came to me: "What if the baudrate is wrong?". Note that I specified the correct baudrate in the debug configuration under Debugger -> Connection. But somehow this does not get passed to gdb.
Under Debugger -> Main, I changed the "GDB debugger" line from `xtensa-lx106-elf-gdb` to `xtensa-lx106-elf-gdb -b 115200` and lo and behold, the connection is established and I can debug my code!

So how exactly is the connection setting passed from sloeber to gdb?

freeck, 2018-11-22:
Well, if you used the example of @Jantje you could see that you missed the function call " gdbstub_init()" in setup().....

Crazor, 2018-11-22:
I tried jantjes examples, watched all relevant videos etc. and faced the same issues every time. My problem is not with the code on the device, as that same code is debuggable just fine when I launch gdb from a shell (or when passing `-b 115200` from inside sloeber). The issue is that sloeber somehow fails to pass the baudrate that I selected in the Debugger settings under Connetion to the gdb instance it launches. As soon as I add that flag, it works.

freeck, 2018-11-22:
Ok, the code you published above is in my opinion not a debuggable....never mind..

The debug plugin of Sloeber is an Eclipse plug not a Sloeber plug in. The baudrate property of a project only effects the download baudrate of Sloeber, not the baudrate of the debug session. So for debugging you have to define the same baudrate in both the setup function as you did, and in the debug-configuration setting. 
I stumbled to this same issue and came to the conclusion that you have to live with it. Or perhaps @jantje knows a work around?

Crazor, 2018-11-22:
I noticed that the two options are not corresponding. But in my case, the option under Debugger -> Connection has no effect at all. I *have* to add the `-b 115200` flag to be able to connect.

Crazor, 2018-11-22:
I've had a look through the upstream issues, but haven't yet found anything related. I will investigate this issue some more on my end and then report it upstream.

Once more, this issue also happens with jantje/hardware, which sent me on my journey in the first place. The solution applies to jantje/hardware, too. As of right now, the debugging example from your video is not working on my end at all until I pass -b 115200 to GDB.

Since jantje/hardware does contain a gdb windows executable, I wonder if the issue is with CDT and the "official" xtensa gdb from the esp8266 Arduino core. How come you distribute your own xtensa-lx106-elf-gdb.exe and not use the one in `packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin`? The only thing I did different while following your video was to choose the xtensa-lx106-elf-gdb from there, since your repo does not contain a mac gdb executable.

Crazor, 2018-11-22:
Oh and can someone check if debugging works on windows when using xtensa-lx106-elf-gdb.exe from packages/esp8266 instead of the one that's shipped with packages/Jantje?

jantje, 2018-11-22:
The root cause of the gnu-mcu is macro expansion. As adding -b 115200 fixes the problem it is very well possible that macro expansion is the problem here as well (different issue, same root cause)
The fact it worked with jantje/hardware and now it no longer works also points into the direction of a cdt change.
Based on the gnu-mcu issue, I'm currently assuming that CDT 9.5.5 fixes this issue. On which CDT version are you? 

>  How come you distribute your own xtensa-lx106-elf-gdb.exe and not use the one in packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin

If I recall correctly, at the time I made the package the gdb executable was not included in the package. I assume esp assumed you can not debug with Arduino IDE and as sutch didn't include it and then saw other tools like Sloeber used to debug so added it later on.
But It is also very well possible I just overlooked the gdb compiler or making the reference to the folder "packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin" is not obvious as there are 2 version dependent folder names.
This is why I included a version in my package by simply copying it. As there was also some cygwin need in windows I added that to.
As long as the parameters don't change you should be able to use the esp8266 provided stuff (if you are not on windows)

jantje, 2018-11-23:
Does this happen with CDT version 9.5.5

jantje, 2018-12-03:
Any feedback on this?

Crazor, 2018-12-04:
I'm sorry, I haven't had the time to investigate this further. But this is on my list for this week! I'll let you know...

Crazor, 2018-12-06:
Sorry again for the delay. I've updated CDT to 9.5.5 and debugging still only works when passing `-b 115200`. Otherwise the debugger does not come up, hanging at "Lauching Test Debug (96%)" and the debugger console showing:

```
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
Ignoring packet error, continuing...
```

jantje, 2018-12-06:
No problems about the delay. We are all busy ...
I hope to have some time to try this on windows this weekend. No promises though.

jantje, 2021-07-11:
As we both didn't get to progressing this issue, I guess this can be closed now?
If not please reopen

closed 2021-07-11 by jantje