Sloeber/arduino-eclipse-plugin #1046: ARDUINO_BOARD symbol build error ESP8266

fow0ryl, 2018-11-24:
I'm getting this issue in a fresh install of Sloeber 4.3.1 or nightly build and ESP8266 2.4.2 on manjaro linux.
It looks like [#493](https://github.com/Sloeber/arduino-eclipse-plugin/issues/493#issuecomment-230584958)
After doing some tests I createt a new project with this code

`#include "Arduino.h"
#include <ESP8266mDNS.h>

void setup()
{
    MDNS.addService("http", "tcp", 80);
}

void loop()
{
}`

First thing to do is to remove automaticaly added include of WiFi lib, since there are conflicting definitions "WiFIUDP" to ESP8266WiFi.

During compile I get this error:
`Building file: /data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/SPI/SPI.cpp
Starting C++ compile
"/data/EclipseX/sloeber//arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/data/EclipseX/sloeber//arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include" "-I/data/EclipseX/sloeber//arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/lwip2/include" "-I/data/EclipseX/sloeber//arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-I/data/EclipseX/Workspace/BUG_Test/Release/core" -c -Wall -Wextra  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=1460   -DARDUINO=10802 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"   -DESP8266   -I"/data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266" -I"/data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/variants/d1_mini" -I"/data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/SPI" -I"/data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266mDNS" -I"/data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src" -MMD -MP -MF"libraries/SPI/SPI.cpp.d" -MT"libraries/SPI/SPI.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/SPI/SPI.cpp"  -o  "libraries/SPI/SPI.cpp.o"
Finished building: /data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/SPI/SPI.cpp
 
Building file: /data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266mDNS/ESP8266mDNS.cpp
Starting C++ compile
"/data/EclipseX/sloeber//arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/data/EclipseX/sloeber//arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include" "-I/data/EclipseX/sloeber//arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/lwip2/include" "-I/data/EclipseX/sloeber//arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-I/data/EclipseX/Workspace/BUG_Test/Release/core" -c -Wall -Wextra  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=1460   -DARDUINO=10802 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_WEMOS_D1MINI"   -DESP8266   -I"/data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266" -I"/data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/variants/d1_mini" -I"/data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/SPI" -I"/data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266mDNS" -I"/data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src" -MMD -MP -MF"libraries/ESP8266mDNS/ESP8266mDNS.cpp.d" -MT"libraries/ESP8266mDNS/ESP8266mDNS.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "/data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266mDNS/ESP8266mDNS.cpp"  -o  "libraries/ESP8266mDNS/ESP8266mDNS.cpp.o"
/data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266mDNS/ESP8266mDNS.cpp: In member function 'void MDNSResponder::enableArduino(uint16_t, bool)':
<command-line>:0:15: error: 'ESP8266_WEMOS_D1MINI' was not declared in this scope
/data/EclipseX/sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266mDNS/ESP8266mDNS.cpp:973:44: note: in expansion of macro 'ARDUINO_BOARD'
   addServiceTxt("arduino", "tcp", "board", ARDUINO_BOARD);
                                            ^
make: *** [libraries/ESP8266mDNS/subdir.mk:20: libraries/ESP8266mDNS/ESP8266mDNS.cpp.o] Fehler 1

10:37:46 Build Failed. 2 errors, 0 warnings. (took 1s.586ms)`

Then I found [#960](https://github.com/Sloeber/arduino-eclipse-plugin/issues/960) and I removed the ESP8266 platform and reinstalled it.
But nothing changed.

At least I changed the defines in  ESP8266mDNS.h and everthing went ok.
`#define ARDUINO_BOARD "ESP8266_WEMOS_D1MINI"`

How to do it the right way?

kkangelvo, 2018-11-24:
I am facing the same issue on Linux Mint 19 with a fresh install of plugin versions 4.2.0, 4.3.0 and 4.3.1.
I tried reinstalling the esp8266 platform with no effect.

Minimal sketch that does not compile:

```C++
#include "Arduino.h"
#include "ESP8266mDNS.h"

void setup(){}
void loop(){}
```

jantje, 2018-11-24:
This is probably related to using " to get strings in your code which is a bad idea as " is used to handle spaces. Like #1028
Modifying the file ../sloeber/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266mDNS/ESP8266mDNS.cpp
`....
addServiceTxt("arduino", "tcp", "board", ARDUINO_BOARD);
...`

to 
```
....
#define VAL(str) #str
#define TOSTRING(str) VAL(str)
....
addServiceTxt("arduino", "tcp", "board", TOSTRING(ARDUINO_BOARD));
```
should do the trick

I can't call this a bug as it (probably works in Arduino IDE) but wanting to make a space, quote and OS safe command line is very challenging.
As there is a way to avoid having to use a quote safe way I would advice to do so.
Unfortunately it is not my call

fow0ryl, 2018-11-24:
I can confirm, that the workaround is working as expected.

Thank your very much. You made a newbie very happy.
(I'm on the way from atmega32 & Windows & Atmel-Studio to linux & arduino & sloeber & ESP right now....)

jantje, 2018-11-24:
Thanks for the feedback
I'm closing this as a duplicate of #1028

closed 2018-11-24 by jantje