Sloeber/arduino-eclipse-plugin #1048: Default list of update sites

sk8nfool, 2018-12-03:
I noticed that CDT is not selected in the list of available update sites - is there a reason for this? I also noticed that there are several un-named sites - what's with them?
 
[image]

jantje, 2018-12-03:
I'm not sure what is going on with the update sites right now but it sure isn't funny.
At my place sites keep on disappearing and names are blanc.
The good news is that if you have the update sites it works :-s
CDT is no longer needed as it is now part of the eclipse stuff
Nebula might be needed
Sloeber nightly or stable ... you know

sk8nfool, 2018-12-03:
CDT integrated into Eclipse. Interesting. Thanks.

uzi18, 2018-12-04:
more like repository of cdt is available from eclipse main repository

closed 2018-12-04 by wimjongman