Sloeber/arduino-eclipse-plugin #1049: Application name on macOS

Crazor, 2018-12-07:
**Describe the bug**
The application name on macOS has cosmetic issues in two places. First, the app bundle is named "sloeber.app" instead of "Sloeber.app", which looks off in the Finder and in the Dock. This is easily fixable for me by renaming, but for consistency, this should be fixed in the build process eventually.

Second, and more important, the name of the main menu (the one in between the  and the File menu) is "Sloeber, the Eclipse Arduino IDE" instead of "Sloeber". This is really taking up unnecessary menu bar space and results in hidden status menus on lower-res monitors like the 13" MacBook Air. I need to switch out of Sloeber to an app with less menus before I can use 1Password, Dropbox etc.
Unfortunately, I didn't find a way yet to patch this on my end.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS Mojave (but all versions affected)
 - Sloeber Version 4.3.1

jantje, 2018-12-07:
As to internationalisation I replaced "Sloeber, the Eclipse Arduino IDE" with"Sloeber"
I however fail to find why it is sloeber.app and not Sloeber.app

wimjongman, 2018-12-10:
Closing. Please reopen if it is not solved.

closed 2018-12-10 by wimjongman

jantje, 2018-12-10:
Thanks @wimjongman :-)