Sloeber/arduino-eclipse-plugin #1054: Sloeber 4.3 how to work with sonoff-tasmota project

MAWMN, 2018-12-19:
Hello, 
  I am using Sloeber 4.3 for programming and flashing my Arduino and Wemos boards.
Is there any description, video or short notes of how to import/use the Sonoff-Tasmota application in(to) the sloeber environment ?
I am using Sloeber plugin 4.3 and the Sonoff Tasmota release 6.4.0

Help will be greatly appreciated.

Thanks in advance..

uzi18, 2018-12-19:
You will not find anything about it, but it is possible to export platformio project for eclipse and import/open it in sloeber

MAWMN, 2018-12-19:
Hello,
thank you for your prompt reply..
Is there any information or example of how to do this, because I am not familiar with PlatformIO

Thanks..

jantje, 2018-12-19:
@MAWMN can you point me to the Sonoff-Tasmota application documentation? More specific on how to use it with arduino?

MAWMN, 2018-12-19:
Hello Jantje,

Global documentation : https://github.com/arendst/Sonoff-Tasmota/wiki

installing it in the Arduino-IDE : https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

regards,

jantje, 2018-12-19:
@MAWMN That would be about the same installation procedure in sloeber.
The differences are the file locations of the installed stuff is in [sloeber]/arduinoPlugin/packages
For the library add the location to the private library options in the preferences.

remember that each time you modified a platform.txt or boards.txt you need to open the project properties and select apply/ok

uzi18, 2018-12-19:
You need only this: https://docs.platformio.org/en/latest/ide/eclipse.html
I have some changes which add some support for tasmota and espeasy, 
but they are not compatible with actual code base of sloeber and it was some. kind of WIP.
Generally platform.io projects has got. files in src and libs in lib dir and sloeber has got some problems with this.

MAWMN, 2018-12-19:
@jantje I will give it a try tomorrow, will let you know the results, or problems I maby will face.

thanks

MAWMN, 2018-12-19:
@uzi18 I presume I will have to install platformIO for this ?

thanks

uzi18, 2018-12-19:
yes, it is used to build tasmota

MAWMN, 2018-12-25:
@ jantje I have ben ill for a couple of days, so the delay..
I am still not able to get my Sonoff project to compliled correctly after trying to to do a lot of testing also.
[image]

Is there anything or perhaps a setting which is causing me these problems ?

jantje, 2018-12-27:
Looking at the image it looks as if you need to upgrade your Sloeber skills before you can handle project that are not fully integrated (yet).
Things that tell me so:
-Nothing on the image states what the problem is.
-There is a folder sonof libraries which (when you follow the sonof instructions should not be there)
-The console output is not visible (here the first error is the interesting one)
-The errors in the makefile are nearly never the root cause nor helping to find the root cause
-The problem view is interesting but it should be at least expanded for errors (and with 100+ errors a screenshot will not do)  

I didn't get this to work but what I did:
Download the zip Sonoff-Tasmota-6.4.1.zip
Extract the zip Sonoff-Tasmota-6.4.1.zip
renamed the folder ..\Sonoff-Tasmota-6.4.1\arduino\version 2.5.0 to ..\Sonoff-Tasmota-6.4.1\arduino\sonof_lib_2.5.0 (spaces are a bad idea in C/C++ world)
[image]
Added this folder to the private libraries in Sloeber
[image]
Created a new project
deleted the created ino file
Dragged and dropped the files from ..\Sonoff-Tasmota-6.4.1\sonoff in the project (selecting copy)
Build (a couple of times).
Now I get this issue (based on the comments there must be more info on the web; I didn't look)
[image]

uzi18, 2018-12-27:
@jantje if you want to compile this it is better to add platformio support

you need this lib: https://github.com/envy/esp-knx-ip

MAWMN, 2018-12-28:
@jantje I was still not able to do an successful compilation, also not after your leads.

Why is it so difficult to get the (an) project working in Sloeber ?

I have done successful compilations and uploads from Arduino IDE, PlatformIO, and VisualStudio-code (with the PlatformIO extension).

I rather would like to use Sloeber, because of all the useful popups etc. you can enable, like the one showing external code when hovering over f.i. a function call. 
This functionality comes in very handy when you have not written the code yourself, to directly see what is happening over there.

MAWMN, 2019-01-04:
I am still struggling to get tasmota sOnOff to compile.

Description	Resource	Path	Location	Type
'AriluxRfDisable' was not declared in this scope	sonoff.ino	/Test2	line 2086	C/C++ Problem
'AriluxRfDisable' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 1516	C/C++ Problem
'CaptivePortal' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 581	C/C++ Problem
'CaptivePortal' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 1827	C/C++ Problem
'class PubSubClient' has no member named 'begin'	xdrv_02_mqtt.ino	/Test2	line 542	C/C++ Problem
'class PubSubClient' has no member named 'lastError'	xdrv_02_mqtt.ino	/Test2	line 566	C/C++ Problem
'class PubSubClient' has no member named 'onMessage'	xdrv_02_mqtt.ino	/Test2	line 546	C/C++ Problem
'class PubSubClient' has no member named 'setOptions'	xdrv_02_mqtt.ino	/Test2	line 544	C/C++ Problem
'class PubSubClient' has no member named 'setWill'	xdrv_02_mqtt.ino	/Test2	line 543	C/C++ Problem
'DhtSetup' was not declared in this scope	sonoff.ino	/Test2	line 2473	C/C++ Problem
'DomoticzSaveSettings' was not declared in this scope	xdrv_07_domoticz.ino	/Test2	line 456	C/C++ Problem
'DomoticzSendKey' was not declared in this scope	sonoff.ino	/Test2	line 1325	C/C++ Problem
'DomoticzSensor' was not declared in this scope	xdrv_03_energy.ino	/Test2	line 633	C/C++ Problem
'DomoticzSensor' was not declared in this scope	xdrv_05_irremote.ino	/Test2	line 165	C/C++ Problem
'DomoticzSensor' was not declared in this scope	xdrv_06_snfbridge.ino	/Test2	line 296	C/C++ Problem
'DomoticzSensorPowerEnergy' was not declared in this scope	xdrv_03_energy.ino	/Test2	line 632	C/C++ Problem
'DomoticzUpdateFanState' was not declared in this scope	sonoff.ino	/Test2	line 397	C/C++ Problem
'DomoticzUpdateFanState' was not declared in this scope	xdrv_02_mqtt.ino	/Test2	line 339	C/C++ Problem
'DomoticzUpdatePowerState' was not declared in this scope	sonoff.ino	/Test2	line 1397	C/C++ Problem
'DomoticzUpdatePowerState' was not declared in this scope	xdrv_04_light.ino	/Test2	line 625	C/C++ Problem
'EnergyMqttShow' was not declared in this scope	xdrv_03_energy.ino	/Test2	line 229	C/C++ Problem
'EnergyMqttShow' was not declared in this scope	xdrv_03_energy.ino	/Test2	line 244	C/C++ Problem
'EnergyMqttShow' was not declared in this scope	xdrv_03_energy.ino	/Test2	line 271	C/C++ Problem
'EnergyMqttShow' was not declared in this scope	xdrv_03_energy.ino	/Test2	line 292	C/C++ Problem
'EnergyMqttShow' was not declared in this scope	xdrv_03_energy.ino	/Test2	line 298	C/C++ Problem
'EnergyShow' was not declared in this scope	xdrv_03_energy.ino	/Test2	line 307	C/C++ Problem
'GetSun' was not declared in this scope	sonoff.ino	/Test2	line 1541	C/C++ Problem
'HandleAjaxConsoleRefresh' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 389	C/C++ Problem
'HandleAjaxStatusRefresh' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 390	C/C++ Problem
'HandleBackupConfiguration' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 399	C/C++ Problem
'HandleConfiguration' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 394	C/C++ Problem
'HandleConsole' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 388	C/C++ Problem
'HandleHttpCommand' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 391	C/C++ Problem
'HandleHueApi' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 1832	C/C++ Problem
'HandleInformation' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 402	C/C++ Problem
'HandleLoggingConfiguration' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 397	C/C++ Problem
'HandleModuleConfiguration' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 395	C/C++ Problem
'HandleNotFound' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 392	C/C++ Problem
'HandleOtherConfiguration' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 398	C/C++ Problem
'HandlePreflightRequest' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 387	C/C++ Problem
'HandleResetConfiguration' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 401	C/C++ Problem
'HandleRestoreConfiguration' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 400	C/C++ Problem
'HandleRoot' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 383	C/C++ Problem
'HandleUpgradeFirmware' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 384	C/C++ Problem
'HandleUpgradeFirmwareStart' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 385	C/C++ Problem
'HandleUploadDone' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 386	C/C++ Problem
'HandleUploadLoop' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 386	C/C++ Problem
'HandleWifiConfiguration' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 396	C/C++ Problem
'HandleWifiConfiguration' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 601	C/C++ Problem
'HAssDiscovery' was not declared in this scope	sonoff.ino	/Test2	line 747	C/C++ Problem
'HueWemoAddHandlers' was not declared in this scope	xdrv_01_webserver.ino	/Test2	line 404	C/C++ Problem
'I2cScan' was not declared in this scope	sonoff.ino	/Test2	line 1278	C/C++ Problem
'LightSerial2Duty' was not declared in this scope	xdrv_04_light.ino	/Test2	line 846	C/C++ Problem
[... cut ...]
this is the location of the previous definition	sloeber.ino.cpp	/Test2	line 12	C/C++ Problem
this is the location of the previous definition	sloeber.ino.cpp	/Test2	line 13	C/C++ Problem
this is the location of the previous definition	sloeber.ino.cpp	/Test2	line 14	C/C++ Problem
this is the location of the previous definition	sloeber.ino.cpp	/Test2	line 15	C/C++ Problem
this is the location of the previous definition	sloeber.ino.cpp	/Test2	line 16	C/C++ Problem
this is the location of the previous definition	sloeber.ino.cpp	/Test2	line 17	C/C++ Problem

uzi18, 2019-01-04:
@jantje can't confirm about build process, but it could be much easier if slober could get src and lib (with priority) dirs support.

uzi18, 2019-01-04:
other problem is about platformio.ini can provide bunch of targets

MAWMN, 2019-01-04:
**The first error of the console..**

..\settings.ino: In function 'void SettingsSaveAll()':
..\settings.ino:233:22: error: 'power' was not declared in this scope
     Settings.power = power;
                      ^

jantje, 2019-01-05:
First off I used the wrong folder as the library folder in my post above. I edited the post. Also the version numbers in the foldername are making things hard for sloeber. After these "fixes" things import really smooth. But still no successful build.

I've investigated in more details and it is .... complicated.
All this complication is caused by ino files and all tools have their own ino to cpp conversion methods.
_As you may know the compiler doesn't know .ino files and the tool/ide needs to convert from .ino files to .cpp files. Arduino does not provide detailed rules on how to do this. Arduino ide cuts and pastes the ino files to one big .cpp file. The consequence of this is that the compiler output needs to be "interpreted" based on the cut and paste work done to generate the .cpp file.
Sloeber makes a  sloeber.ino.cpp file that includes the .ino files. This avoids the need of the interpretation of the build output but it is "limiting the posibilities"_
In other words: **Sloeber support for ino files is only intended to help create library examples. It is clear we hit some limitations of Sloeber here.**
The sonof project clearly uses ino files to it's full extend and this is where it gets hard.

The first thing that hit me was that (because of dragging and dropping the code) the main ino file did not have the same name as the project. This is important because the order of processing ino files is crucial (by design so not a sloeber issue).
To fix simply rename sonof.ino to [myProjectName].ino

The second thing that hit me is related to the fact that the Arduino IDE does not allow you to define stuff at the project level. The sonoff-tasmota people worked around this by using a my_user_config.h which contains lots of defines that (de)activate parts of the code.
Due to the ino format (here the concatenation of all ino files to one big file) the indexer (and as such Sloeber) gets confused because most of the ino files do not contain
#include "my_user_config.h"
So sloeber does not know which code needs to be "included".
I opted to add #include "my_user_config.h" at the top of all the .ino files which made things "better" 

Then I got hit by
.```
.\sloeber.ino.cpp:183:8: error: 'bmp_sensors_t' does not name a type
 extern bmp_sensors_t* bmp_sensors;
        ^
..\sloeber.ino.cpp:200:8: error: 'bmp180_cal_data_t' does not name a type
 extern bmp180_cal_data_t* bmp180_cal_data;
        ^
..\sloeber.ino.cpp:225:8: error: 'Bme280CalibrationData_t' does not name a type
 extern Bme280CalibrationData_t* Bme280CalibrationData;
```
 bmp_sensors is defined in xsns_09_bmp.ino as
```
typedef struct {
  uint8_t bmp_address;    // I2C bus address
  char bmp_name[7];       // Sensor name - "BMPXXX"
  uint8_t bmp_type;
  uint8_t bmp_model;
#ifdef USE_BME680
  uint8_t bme680_state;
  float bmp_gas_resistance;
#endif  // USE_BME680
  float bmp_temperature;
  float bmp_pressure;
  float bmp_humidity;
} bmp_sensors_t;
```
And here we get hit by a variant of #775. In this case it is not a method but a global variable.

jantje, 2019-01-05:
I got it to compile. Can't test the outcome as I don't know what this does.
It is not to bad but due to "very hard to sell these changes to the provider" I'm not sure whether this is a route you want to pursue.
Anyway I needed to add #include "sonoff.h" to my_user_config.h (I did it at the top)
I made a new file called xsns_09_bmp.h with following content 
```
#include "Arduino.h"
#include "my_user_config.h"

typedef struct {
  uint8_t bmp_address;    // I2C bus address
  char bmp_name[7];       // Sensor name - "BMPXXX"
  uint8_t bmp_type;
  uint8_t bmp_model;
#ifdef USE_BME680
  uint8_t bme680_state;
  float bmp_gas_resistance;
#endif  // USE_BME680
  float bmp_temperature;
  float bmp_pressure;
  float bmp_humidity;
} bmp_sensors_t;

typedef struct {
  int16_t  cal_ac1;
  int16_t  cal_ac2;
  int16_t  cal_ac3;
  int16_t  cal_b1;
  int16_t  cal_b2;
  int16_t  cal_mc;
  int16_t  cal_md;
  uint16_t cal_ac4;
  uint16_t cal_ac5;
  uint16_t cal_ac6;
} bmp180_cal_data_t;

bmp180_cal_data_t *bmp180_cal_data = NULL;

typedef struct {
  uint16_t dig_T1;
  int16_t  dig_T2;
  int16_t  dig_T3;
  uint16_t dig_P1;
  int16_t  dig_P2;
  int16_t  dig_P3;
  int16_t  dig_P4;
  int16_t  dig_P5;
  int16_t  dig_P6;
  int16_t  dig_P7;
  int16_t  dig_P8;
  int16_t  dig_P9;
  int16_t  dig_H2;
  int16_t  dig_H4;
  int16_t  dig_H5;
  uint8_t  dig_H1;
  uint8_t  dig_H3;
  int8_t   dig_H6;
} Bme280CalibrationData_t;

Bme280CalibrationData_t *Bme280CalibrationData = NULL;

#pragma once
```
I removed the declarations above from xsns_09_bmp.ino and added #include "xsns_09_bmp.h" at the top

I added a #pragma once to Adafruit_SGP30.h

I may have done some more trivial stuff I forgot. The indexer may hickup. If you still get undefines rebuild the index (wait for it to finish) and rebuild.
I also increased the memory usage of Sloeber because this is a big project (embedded wise speaking) and I think the indexer was getting memory restrained. To do so mod the  sloeber.ino file (next to sloeber.exe) 
I did 4 times more (I have plenty of memory) 2 times is probably sufficient
```
-Xms160m
-Xmx2048m

```

jantje, 2019-01-07:
As this was a question I consider the question answered

closed 2019-01-07 by jantje

uzi18, 2019-01-07:
@jantje I mean you can define several targets in platformio.ini and build once all these targets or choose one

jantje, 2019-01-07:
@uzi18 you can do so in sloeber as well. I use this to setup debug stuff separate from the "release" stuff.
It is called configurations. Sloeber does not support building al configurations in one go. However I don't really see this as a major constraint.

uzi18, 2019-01-08:
@jantje did not tested this feature, but in platformio.ini you can set some defines like -DSOMEFEATURE=1

jantje, 2019-01-08:
if you go in project properties->arduino you can  do so as well. Did you watch any of the video's I made? You can find them via sloeber->menu->arduino

tommyjlong, 2019-02-26:
Hi,
FYI, I pretty much followed the directions and was able to get all the modules to compile, but ran into a problem with the linker not having enough room in iram1_0.  I did some comparisons of the sonoff.map file created by Sloeber versus one created by Arduino IDE but nothing obvious.  By removing the IR feature (by commenting out USE_IR_REMOTE and USE_IR_RECEIVE in my_user_config.h) I reduced the iram1_0 enough to get it to link successfully.  Not sure why, but there are a few hundred bytes difference between what Arduino IDE uses vs. Sloeber.  If I get a chance later, I'll redo this from scratch and see if I get the same results.

I did load this up on a nodeMCU board and have WiFi and Web pages working. It was a bit of a quick test, with the only problem thus far is mDNS doesn't appear to be working.

Anyway, first time user with Sloeber and thanks for all the work on it!!

uzi18, 2019-02-27:
It is normally separated build, because of lot iram use in ir/irtx plugins.
Nice try.

Thanks for hint maybe we can reduce iram usage a little.

greg-ware, 2019-03-12:
@jantje , @uzi18 , @MAWMN , I was about to embark on the same, Tasmota is a very fine piece of software engineering but trying to modify code with PlatformIO is a hurdle.
So, thanks for this thread, it will be useful to me too...

executer-uno, 2019-12-20:
> I got it to compile. Can't test the outcome as I don't know what this does.
> It is not to bad but due to "very hard to sell these changes to the provider" I'm not sure whether this is a route you [...]

Hi, Im also in trouble to compile external library with Sloeber.
Do You have any idea, how did you rid of "-Werror=unused-value" errors?