Sloeber/arduino-eclipse-plugin #1058: Issue to compile ESP8266 projects

painlessthetrue, 2019-01-04:
Dears,

I´m not able to compile any ESP8266 project. Using Sloeber 4.3 last stable version and with avr-boards everyhing is working fine. 
After one day of searching, i give up and like to ask for help:

I use ESP8266 platform V2.4.1, evenmore without any success i tried 2.5.0 beta2 too. It´s not depennding from selected board and version...

The compiler-states are strange and it seems to be a problem with the path/enviroment delimiter during construct of makefiles.
Here a extract output from subdir.mk:

_# Each subdirectory must supply rules for building sources it contributes
sloeber.ino.cpp.o: ..\sloeber.ino.cpp
	@echo 'Building file: $<'
	@echo 'Starting C++ compile'
	${A.RECIPE.HOOKS.SKETCH.PREBUILD.1.PATTERN}
		"C:\s\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\s\/arduinoPlugin/packages_/esp8266/hardware/esp8266/2.4.1/tools/sdk/include" "-IC:\s\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/lwip/include" "-IC:\s\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IZ:/WS1/t/Release/core" -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DARDUINO=10802 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="\"ESP8266_NODEMCU\""   -DESP8266   -I"C:\s\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266" -I"C:\s\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\variants\nodemcu" -MMD -MP -MF"$(@:%.o=%.d)" -MT"$(@)" -D__IN_ECLIPSE__=1 -x c++ "$<"  -o  "$@"__

	@echo 'Finished building: $<'
	@echo ' '
  
---> C:\s\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++<<<<

is not a valid windows path. Mixing of  **\ and /**

I try to figure it out from project-properties\C/C++ Build/Enviroment and found the variable
A.COMPILER.PATH with the value ${A.RUNTIME.TOOLS.XTENSA-LX106-ELF-GCC.PATH}/bin/
Here seems to be the wrong join of compliler commands.

In the output windows is something wrong too:
make -C "C:\s\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/tools/sdk/lwip/src" install TOOLS_PATH="C:\s\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/**xtensa-lx106-elf-**"

The rest of progam name is missing...

I know the restrictions of path variables in windows and tried to shorten them so far as possible.

Thx in advance for any reply !

painless

jantje, 2019-01-05:
ESP8266 is working fine with Sloeber. There are many users of this hardware. I'm one of them. However version 2.5.0 beta2 doesn't work properly. As it is a beta I didn't look into it 

>is not a valid windows path. Mixing of \ and /

windows can handle that. It is not the problem of your issue

I guess you are using the option build wip from source and you do not have the tools on your system to build it. Please try to build using a wip build version

painlessthetrue, 2019-01-05:
Perfect. I got it. The solution is already documented under 
#927 
https://github.com/Sloeber/arduino-eclipse-plugin/issues/927

Thy for your support !

closed 2019-01-05 by painlessthetrue